Ultimaker/Cura #8715: [4.7.1] ChangeAtZ script Extruder Temp changes reverted on Z hop

Hirsute, 2020-11-09:
**Application version**
4.7.1

**Platform**
Windows 10, NVidia GForceRTX 2070

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Load Model
2. Turn on Z-hop When Retracted
3. Setup ChangeAtZ to change Extruder 1 temp at Height
4. Slice, save gcode to file. 
5. ChangeAtZ sets Extruder Temp correctly, but then almost immediately sets it back to the Default Print Temp (see snippet below and attached gcode files)

**Screenshot(s)**
N/A - Code File more helpful.

**Actual results**
`
;[CAZD:
M104 S240.0 T0
;:CAZD]
G1 F300 Z28
G0 F9000 X99.335 Y113.52 Z28
G0 X89.985 Y105
G1 F300 Z28
G0 F9000 X99.945 Y110.489
;TYPE:SKIN
;[CAZD:
M104 S260.0 T0
;:CAZD]
`

**Expected results**
`;[CAZD:
M104 S240.0 T0
;:CAZD]
`

**Project file**
Project Files and GCode Files zipped and attached.

**Log file**
cura.log attached.

**Additional information**
I've tried with multiple profiles, including a stock profile with the only change being turning on Z-Hop When Retracted and Setting up ChangeAtZ
[cura.log](https://github.com/Ultimaker/Cura/files/5508204/cura.log)
[CAZ-Extruder-Temp-Z-Hop-Issue.zip](https://github.com/Ultimaker/Cura/files/5508209/CAZ-Extruder-Temp-Z-Hop-Issue.zip)

Ghostkeeper, 2020-11-17:
Yeah, I could reproduce this issue. It seems to mis-detect what a layer's height is since it already triggers when the Z coordinate first reaches the specified Z coordinate, even though the actual layer is still a bit below that height; it only triggers because of the Z hop.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-24 by github-actions[bot]