Ultimaker/Cura #8719: Print with Octoprint issue

Nickelindgren, 2020-11-09:
**Application version**
4.7.1

**Platform**
Win10

**Printer**
Creality Ender 3 PRO

**Reproduction steps**
1. Sliced and sent to octoprint

**Actual results**
Sometimes my extruder starts feeding backwards right after completing the ABL and just when it's about to print the first line. 

**Expected results**
It should print without any issues

rburema, 2020-11-09:
Can you attach the generated gcode, and the logs?

dxgldotorg, 2020-11-10:
Does this occur when you save the gcode to SD card, or save to file then upload to OctoPrint via browser?

Nickelindgren, 2020-11-10:
Now I'm not so sure this is an bug in the cura software. I discovered that I sometimes got the same issue when sending a gcode straight to octoprint and not through cura. I will close this issue since the problem seams to be elsewere.

closed 2020-11-10 by Nickelindgren