Ultimaker/Cura #8720: Inadequate support when model is rotated using Auto-Orientation extension Cura 4.8

mereseydotes, 2020-11-09:
**Application version**
4.8

**Platform**
Windows 10 64 bit

**Printer**
Creality CR-6 SE

**Reproduction steps**
1. Import model (I will attach GCODE)
2. Use Auto-Orientation extension
3. Slice

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
So, when I had overhang for supports set to 60, I got way too few supports. When I changed it to 45, there's only missing supports for one of the dog's hind legs. When I go with the original imported orientation, it looks as if the support is adequate, but I haven't printed yet. It's hard to tell from the Preview.

**Expected results**
I would hope there would be support wherever needed

**Project file**
Here's a link to what I hope are the relevant files:
https://drive.google.com/file/d/11hEuJBSdyDirvbOa8UIDbbSIUFru-fxH/view?usp=sharing

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-11-10:
In you attached gcode, I can see the problem, however, I have the release candidate for 4.8 that is very close to what we're going to release within the next 24 hours, and I can't seem to reproduce it there:

[image]

mereseydotes, 2020-11-10:
Were you able to reproduce it on the beta version? Has the support calculation changed from the beta version? And also, is it OK for me to remove the files from my Google drive? Thanks!

https://github.com/Ultimaker/Cura/issues/8720#issuecomment-724545286

rburema, 2020-11-10:
@mereseydotes Can you try the 4.8.0 stable that is out today?

(I haven't tried the actual beta with this, as I had only the newer build installed.)

mereseydotes, 2020-11-10:
Oh. I guess I'll just trust that it's fixed, rather  than just not a problem on your installation. Is it OK for me to delete the files?

https://github.com/Ultimaker/Cura/issues/8720#issuecomment-724822733

rburema, 2020-11-10:
I've saved the files locally, the other devs know how to contact me, you can delete them :-)

rburema, 2020-11-10:
I'm closing this now, but please post a reply to this if it re-appears please post a message to this issue.

closed 2020-11-10 by rburema