Ultimaker/Cura #8722: [4.8 beta] Compensate Inner Wall Overlaps seems inconsistend

thelastWallE, 2020-11-10:
**Application version**
4.8 beta

**Platform**
win 10

**Printer**
ender 3

wall gets not compensated:
layer 34 https://i.imgur.com/JhKG7oN.png

wall gets compensated:
layer 37 https://i.imgur.com/rz7BRH3.png

**Actual results**
inner walls gets no compensation at some places. I also tried to change minimum wall flow up and down to see if the one wall would be removed but there was no compensation at all.

**Expected results**
inner wall reduced or removed(minimum wall flow) next to other inner wall

**Project file**
[cube_layer34_innerwallOntheX.zip](https://github.com/Ultimaker/Cura/files/5515995/cube_layer34_innerwallOntheX.zip)

**Additional information**
I know that there is a new function worked on for adaptive line width. But i think this is also a strange behavior from the existing compensation function that it may be worthy to look at.
Maybe the new function can have a bug on these places, too. There it just will not use the variable line width, too.

Ghostkeeper, 2020-11-17:
Indeed I'd expect that the wall gets compensated in the first picture there at layer 34. The bug is reproduced with that project file. I'll put this on our pin board to discuss tomorrow to estimate how difficult this would be to fix.

Ghostkeeper, 2020-11-19:
We've decided to defer this issue. The most major reason for that is that the Wall Overlap Compensation is on its way out due to a major change we're making in the way we generate walls: Ultimaker/CuraEngine#1210. That PR would remove this issue as well. The change is not expected for the upcoming 4.9 release but perhaps the release after that.

Ghostkeeper, 2021-12-14:
This is no longer relevant due to https://github.com/Ultimaker/CuraEngine/pull/1210 being merged now.

closed 2021-12-14 by Ghostkeeper