Ultimaker/Cura #8724: [4.8] infill for walls seems broken

7FM, 2020-11-10:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
Slice attached STL with 30% Infill and Cubic Pattern, Wall Thickness 0.2, Wall Line Count 2
(feel free to ask for more relevant profile settings, or is there an easy way of exporting?)

**Screenshot(s)**
Created Infill: 
[image]
Same Layer but after printing infill: 
[image]
Same STL & Profile sliced with Cura 4.6.2 looks better: 
[image]

**Actual results**
Inner wall lines are printed with infill as well as solid in a single layer.

**Expected results**
Per layer the walls should either be printed with infill or solid per line.

**Project file**
[project_file.zip](https://github.com/Ultimaker/Cura/files/5519255/project_file.zip)

jellespijker, 2020-11-10:
I can't reproduce it on master, I suspect it was fixed with ultimaker/curaengine#1353
[image]

shauneccles, 2020-11-14:
Confirming this bug in shipped 4.8.

konskarm, 2020-11-16:
I can reproduce this. What you are seeing here is the gap filling technique overlapping with the infill. 

**Fill Gaps Between Walls: Everywhere**
[image]

**Fill Gaps Between Walls: Nowhere**
[image]

This is something we also recently discovered. However, we decided we won't fix that in the current master, since it will be fixed by the new [variable-line width, wall-generating technique](https://github.com/Ultimaker/CuraEngine/pull/1210) that we have in the works.

konskarm, 2020-11-16:
You can work around this issue for the time being by increasing the Wall Line Count by 1 and keeping the Fill Gaps Between Walls as Everywhere. Then the small gap will be filled only by the gap-fillling but not by infill. This happens because CuraEngine tries to add an extra wall to replace the infill, but the wall doesn't fit, so it is then replaced by a gap-filling (yellow) wall with smaller width.

[image]

As a small preview, this is how it will look like using libArachne:
[image]

The middle inner wall, which cannot be as wide as the rest of the inner walls, is properly printed with an adapted line width.

7FM, 2020-11-16:
> As a small preview, this is how it will look like using libArachne:
image
The middle inner wall, which cannot be as wide as the rest of the inner walls, is properly printed with an adapted line width.

On the image you have increased the wall count to 3 (see circle) too which is your current suggested workaround, I hope it works with a wall count of one less as well. Else the bug would still be present.

Besides that I am not sure if the suggested workaround will always work, because increasing the wall count might cause new problem regions as previously there was enough free space that wall gab filling wasn't triggered (but is triggered with increased wall count causing the problem). If so you would have to increase the wall count as long as this problem occurs, resulting in longer prints & filament usage (if needed I might be able to create a proof of concept showing this hypothesis). Maybe setting fill gabs between walls to nowhere is a better workaround.

konskarm, 2020-11-16:
That's true. We haven't implemented the gap-filling technique to work with libArachne yet, so the middle wall at the moment will not be filled at all. Nevertheless, it will look something like the above image, with the difference that the middle wall will be a skin wall (since infill cannot fit in there).

Ghostkeeper, 2021-12-14:
The original issue here was fixed with Cura 4.12. And then the gap filling is entirely replaced by https://github.com/Ultimaker/CuraEngine/pull/1210 too.

closed 2021-12-14 by Ghostkeeper