Ultimaker/Cura #8726: Support for multiple cores in OS X

GJSchaller, 2020-11-10:
After realizing Cura 4.8 Beta was not using the full processing power of my MacBook Pro, I did a bit of research and came across this closed bug: https://github.com/Ultimaker/Cura/issues/7173

It notes that Cura for OS X does not support multi-threading.  Is this a feature that can be added?

I know OS X has some features that might help with this, such as Metal (although that's mostly for graphical rendering, it may apply to Cura as well)

Ghostkeeper, 2020-11-17:
The reason why this doesn't work is because we are using OpenMP to implement the multi-threading in CuraEngine. OpenMP is not supported by the Apple Clang compiler: https://mac.r-project.org/openmp/
There are no known solutions to this except for highly experimental hacks.

Instead of those hacks we'd be better off using a different compiler on MacOS. But that is a major change to our build system which is difficult and dangerous as long as the Cura team has no real DevOps engineer and no developer programming on MacOS. (We have a MacOS computer for testing and that tester does some devops as well, but it's a bit hard to work with.)

Those are the reasons why MacOS is slicing single-threaded right now.

GJSchaller, 2020-12-19:
Thank you for the information - this makes absolute sense.  Is there anything we, as the macOS community, can do to assist?

smartavionics, 2020-12-19:
Hi, I may be mistaken because I did it a long time ago and I don't actually use MacOS myself but I believe that I enabled multi-core support in my MacOS Cura builds. You can find my most recent build at https://github.com/smartavionics/Cura/releases/tag/20201121 so maybe you could try that and see if it does  actually use multiple cores.

smartavionics, 2020-12-19:
Yes, I did enable multi-core support, the README.md file says this...

> 20190917 - Builds for MacOS 10.13 now available. These use multi-threading in CuraEngine and so should slice quicker than the upstream releases.

GJSchaller, 2020-12-19:
> Yes, I did enable multi-core support, the README.md file says this...

Unfortunately, your fork seems to have killed Cura altogether on my system.  I tried to launch your build, and it wouldn't display the UI at all - I trashed the app, and tried to launch Cura 4.8, which now comes up to a blank gray screen and is unable to be interacted with.  I can't see my preferences, printers, or anything else - only blank gray screens.

I've tried removing the official build and any support files from macOS, but the issue is persisting.

smartavionics, 2020-12-20:
Hmm, I'm very sorry about that. It's not a problem that I have heard of before. Please try removing ~/Library/Application Support/cura/master/cache and maybe that will help. BTW, what version of MacOS are you using? My builds don't yet run on Big Sur but work OK (AFAIK) on other MacOS releases.

smartavionics, 2020-12-20:
Also, and more likely to be beneficial, you should remove ~/Library/Caches/Ultimaker\ B.V./Ultimaker\ Cura\qmlcache

smartavionics, 2020-12-20:
I just tried it on my ancient MacBook that has a 2 core processor and CuraEngine was using around 180% of CPU so I guess the multi-core slicing works OK. Hope you can get the UI functional again!

GJSchaller, 2020-12-20:
I'm running Big Sur - the issue is persisting across multiple user accounts, so it's not an issue local to the user.

I'm going to back up & wipe my MacBookPro - at this point, it's due for a clean install anyways.

soostdijck, 2021-05-20:
This is still a problem on Big Sur 11.3.1 and Cura 4.9.1. Which is really a shame, slicing is slow and my CPU is just sitting there... It's using two threads only.

GJSchaller, 2021-12-21:
Don't know if this will help or not, but Blender just got some major support from Apple - it may be possible to court them for some help, given how big of an industry 3D printing is becoming.

https://www.macrumors.com/2021/12/15/blender-metal-gpu-rendering-m1-macs/

pinguinpfleger, 2022-01-26:
any news on this?

Ghostkeeper, 2022-02-02:
Somewhat. We have updated to C++17 which makes the multi-threaded algorithms of the STD library available. However we're finding that Apple Clang also doesn't support those: https://en.cppreference.com/w/cpp/compiler_support

(Look for "Parallelism TS" on that page.)

soostdijck, 2022-02-03:
I would even consider shifting any efforts to looking at M1 cpus instead, the intel macs are on their way out

Ghostkeeper, 2022-02-17:
On their way out maybe, but still the majority of our MacOS users. We tend to base decisions about what we need to support on the usage statistics we get from slices. Dropping support for 60% of our MacOS users is not going to go over well. We'll consider dropping Intel support for Mac if that drops to just a few percent, I'd reckon.

Qt 6.2 (planned for the next release) is the first Qt release that actually says it supports the M1 CPU. We intend to ship the x86_64 binaries still, which should then get emulated well by Rosetta too on the M1, supporting both.

soostdijck, 2022-02-19:
So "dropping support" suggests that there actually is support. We're still only using one core as far as I know?

GJSchaller, 2022-02-21:
> So "dropping support" suggests that there actually is support. We're still only using one core as far as I know?

We're still on one core.  The original reason given was the compiler used didn't support multi-core functionality, and in the process of moving to a new compiler, M1 support was brought up.  There's no reason both can't be accomplished by the move, but I do agree that multi-core support is the main issue that needs to be addressed.

nallath, 2022-02-21:
Well, the new compiler _also_ doesn't properly support standard C++17 stuff. Unfortunately developing stuff for OSX is a hot mess :( 

As for the support that blender got, it's about GPU rendering. What we need is multithreaded CPU support.

Ghostkeeper, 2022-03-07:
> So "dropping support" suggests that there actually is support. We're still only using one core as far as I know?

This was a response to "intel macs are on their way out". Yes, we currently support the Intel chips on MacOS, although it will indeed slice with only 1 core.

GJSchaller, 2022-06-03:
Cura 5.0 is out, but still not using the full potential of my Mac's multi-core processor.  Any chance this can be looked at?

GJSchaller, 2022-07-22:
Looks like 5.1 supports this... I think?  I see it in the notes, but my CPU usage isn't that high.

GJSchaller, 2022-07-22:
Never mind, it works!  Activity Monitor confirms it!