Ultimaker/Cura #8727: [4.8.0] Unable to open files.

jonwienke, 2020-11-10:
**Application version**
4.8.0

**Platform**
Windows 10, latest updates.

**Printer**
CrealityEnder 5+

**Reproduction steps**
I just uninstalled 4.7.1 and installed 4.8.0. I am now unable to open any file, either with the open button, open menu, open recent menu, or drag/drop. It doesn't matter whether the file is local or cloud. It also doesn't matter whether it's a project, STL, or gcode file.

**Screenshot(s)**
[image]

**Actual results**
Nothing. The program completely ignores the open file command. If I try to open a gcode file and have already attempted to do so, I get the error popup shown above. But for any other file type, the open dialog disappears after clicking OK, but nothing happens after that.

**Expected results**
The file(s) should open.

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

jonwienke, 2020-11-11:
Also, after attempting to load a gcode file, trying to open any other filetype triggers this error:
[image]

nallath, 2020-11-11:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

Please don't forget to include log files.

jonwienke, 2020-11-11:
Here's the log:
[cura.log](https://github.com/Ultimaker/Cura/files/5525549/cura.log)

nallath, 2020-11-12:
I can only see that you try to load a 3mf file, which seems to be corrupt.

jonwienke, 2020-11-12:
You didn't look very hard, here's two load attempts:
2020-11-10 17:46:59,043 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-11-10 17:47:28,451 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/jonwi/OneDrive/CNC/Projects/YarnBowl/YarnBowl.stl
2020-11-10 17:47:47,644 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-11-10 17:47:57,706 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/jonwi/OneDrive/CNC/Projects/YarnBowl/YarnBowl Lid.stl
2020-11-10 17:48:11,642 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

I saved the 3mf file after loading some STLs with no result, then tried loading it back to see what happened. File size is zero bytes, or I would post it.

nallath, 2020-11-12:
Could you also share those files then? So far i've not been able to reproduce your issue.

jonwienke, 2020-11-12:
I can, but they open fine in Creality Slicer and other programs.
[YarnBowl.zip](https://github.com/Ultimaker/Cura/files/5530868/YarnBowl.zip)

jonwienke, 2020-11-16:
Update:
I left Cura open overnight after attempting to open an STL, and in the morning, it had the file open. And it seems to be working normally ever since. So something odd is intermittently delaying the opening of files for a period of hours.

[cura.log](https://github.com/Ultimaker/Cura/files/5548908/cura.log)

Ghostkeeper, 2020-11-17:
I think the problem derives from this part in the log, a bit before the STLs are being loaded:

```
2020-11-10 18:38:31,418 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/jonwi/OneDrive/CNC/Projects/Death Whistle/Mayan_Death_Whistle_90.gcode
2020-11-10 18:38:31,421 - INFO - [MainThread] UM.Controller.deleteAllNodesWithMeshData [468]: Clearing scene
2020-11-10 18:38:31,424 - INFO - [MainThread] cura.CuraApplication._removeNodesWithLayerData [1995]: Clearing scene
2020-11-10 18:38:51,342 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/jonwi/OneDrive/CNC/Projects/Death Whistle/Mayan_Death_Whistle_90.gcode
2020-11-10 18:41:26,128 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/jonwi/OneDrive/CNC/Projects/Death Whistle/Mayan_Death_Whistle_90.gcode
2020-11-10 19:14:12,582 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [130]: Writing to [C:/Users/jonwi/OneDrive/CNC/Projects/YarnBowl/test.3mf]...
2020-11-10 19:14:12,584 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [114]: 'info/send_slice_info' is turned off.
2020-11-10 19:14:12,585 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [151]: Writing to Local File C:/Users/jonwi/OneDrive/CNC/Projects/YarnBowl/test.3mf in binary mode
2020-11-10 19:14:22,598 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-11-10 19:14:42,591 - DEBUG - [MainThread] UM.Message._onInactivityTriggered [101]: Hiding message because of inactivity
2020-11-10 19:14:50,610 - INFO - [MainThread] cura.CuraApplication.reloadAll [1429]: Reloading all loaded mesh data.
2020-11-10 19:14:59,172 - INFO - [MainThread] UM.Controller.deleteAllNodesWithMeshData [468]: Clearing scene
```

So you can see that the .gcode file is starting to load 3 times, but never does it actually come down to even parsing the file. To compare, this is what happens when I load in a .gcode file:

```
2020-11-17 17:00:49,472 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///home/trin/Gedeeld/Projects/SlicerTestModels/test.gcode
2020-11-17 17:00:49,473 - INFO - [MainThread] UM.Controller.deleteAllNodesWithMeshData [468]: Clearing scene
2020-11-17 17:00:49,474 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=1) <UM.Operations.RemoveSceneNodeOperation.RemoveSceneNodeOperation object at 0x7f970a4f1520>, took 0ms
2020-11-17 17:00:49,475 - INFO - [MainThread] cura.CuraApplication._removeNodesWithLayerData [1995]: Clearing scene
2020-11-17 17:00:49,510 - DEBUG - [Thread-7] GCodeReader.FlavorParser.processGCodeStream [315]: Preparing to load g-code
2020-11-17 17:00:49,539 - DEBUG - [Thread-7] GCodeReader.FlavorParser.processGCodeStream [355]: Parsing g-code...
```

So I think the problem is more likely inside that g-code file. Perhaps it's not able to read from that OneDrive where the g-code file is loaded?
Since it appears to work well after a few hours, perhaps that .gcode file is just so big that it takes a few hours to download/upload from/to the OneDrive? I'm not really sure how that OneDrive works actually. For instance Dropbox would save it to disk first and then slowly upload in the background, and slowly download in the background before actually opening the file in Cura, so it would most likely not suffer this issue.

jonwienke, 2020-11-17:
The death whistle gcode file is not large, is mirrored locally, and I've printed it without incident over a dozen times. Upload and download time is a few seconds. It's printing the model at 90% normal scale, so slightly smaller than default.

Ghostkeeper, 2020-11-17:
Yeah that shouldn't be a problem then. Maybe there is an error that we're hiding unintentionally.

Ghostkeeper, 2020-11-19:
I'm still unable to reproduce this issue so far. We'll keep an eye out if there are more reports about this. Maybe we can find a commonality between your cases then.

martan3d, 2021-06-05:
An old thread but it just happened to me.  Cura quit loading files.  Upgraded to 4.9.1. Same. Looks like USB discovery issue perhaps?  I have an AVR programmer, an RPI, and an Xbee carrier plugged in to my USB extender.  I could see in the log files it was taking a long time to look at all the COM ports.  Unplugged them and magic.  It all worked.  Restarted Cura, it loaded the file right away.

Ghostkeeper, 2021-06-14:
> Cura quit loading files.

You have a different problem then. This issue is about it showing a message that you can't open multiple g-code files at the same time even if you load other types of files. Please refer to https://github.com/Ultimaker/Cura/issues/9668 instead.

closed 2024-11-20 by GregValiant