Ultimaker/Cura #8729: Toggle for plate disallowed area

Sprint8, 2020-11-11:
Please for the love of Cura find a way to toggle the disallowed space on the Ender 3 hotbed and probably all others as I use the whole space.  

My alternative was to make the area smaller because I use the small 12.7mm clips but then why not have an edit feature or just the ability to turn it off. Then I wouldn't have to manually edit the definition every time there's a new release.

Thanks.

Ghostkeeper, 2020-11-17:
There is the Ender 3 Pro definition which is the exact same as Ender 3, but without the disallowed areas since that Pro edition apparently has a magnetic build plate instead of clips. It also has an adjusted build volume because of that but you can change that in the Machine Settings in the interface if you like.

The disallowed areas have been added in Cura 4.5 from this discussion: https://github.com/Ultimaker/Cura/issues/6267
We won't just revert that decision willy-nilly. The disallowed areas were added by the community after deliberation. I don't have that printer, so I can't make decisions for it. If the community wants to make a change, they'll need to change it themselves through a pull request. In this instance since the topic has had some heated debate, we'll need more people weighing in on it too.

closed 2020-11-17 by Ghostkeeper

Sprint8, 2020-11-17:
I didn't realize this was such a hotbed issue.   Thanks for the detailed explanation.  I will just do my own thing :)

nippoo, 2021-01-14:
I had the same issue - thanks for the Ender 3 Pro tip. (Sorry for the spam - I'm mainly commenting to acknowledge @Sprint8's  _excellent_ "hotbed issue" pun...)