Ultimaker/Cura #8730: [2.1] Unable to go through first time setup on KDE

nils-van-zuijlen, 2020-11-11:
**Application version**
2.1
Also seen compiling from master (https://github.com/Ultimaker/Cura/commit/9afbc64934fc7e4c2e14ab516938e6a935e48336)

**Platform**
Linux, Archlinux, KDE Plasma 5.20.3
Installed from community or aur

**Printer**
None...

**Reproduction steps**
1. Launch Cura for the first time (without any configuration)
2. That's it!

**Screenshot(s)**
[image]

**Actual results**
Cura starts and prompts to start the setup, but no button is visible nor clickable (I tried clicking almost anywhere on screen without any change)

**Expected results**
A button to continue setup should be visible

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5523088/cura.log)

Most relevant part (I think):
```
2020-11-11 10:10:36,300 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [408]: file:///usr/lib/qt/qml/QtQuick/Controls.2/org.kde.desktop/TabBar.qml:30: TypeError: Cannot read property 'height' of null
2020-11-11 10:10:36,303 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [408]: file:///usr/share/cura/resources/qml/Menus/ConfigurationMenu/CustomConfiguration.qml:233: TypeError: Cannot read property 'isEnabled' of null
2020-11-11 10:10:36,308 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [408]: file:///usr/share/cura/resources/qml/Cura.qml:410: TypeError: Cannot read property 'Component' of null
2020-11-11 10:10:36,331 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [408]: file:///usr/lib/qt/qml/QtQuick/Controls.2/org.kde.desktop/TabBar.qml:30: TypeError: Cannot read property 'height' of null
2020-11-11 10:10:36,333 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [408]: file:///usr/share/cura/resources/qml/PrintSetupSelector/Custom/MenuButton.qml:42:18: QML Label: Cannot anchor to an item that isn't a parent or sibling.
2020-11-11 10:10:36,336 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [408]: file:///usr/lib/qt/qml/QtQuick/Controls.2/org.kde.desktop/ComboBox.qml:96:9: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
```

**Additional information**
From the logs, it seems to be caused by a name clash between Cura and KDE Qml definitions.

Also when closing, Cura segfaults.

fieldOfView, 2020-11-11:
If this is Cura 2.1, then I am Micky Mouse ;-)

Only the AppImage as distributed by Ultimaker is official supported here. It comes with the dependencies that it is known to work with.

jellespijker, 2020-11-11:
duplicate of #8472

This because you probably have a newer Qt5 version installed system-wide then Cura supports (5.10). That is because the AUR builds it from source. I personally use Manjaro and the AUR package and I came across the same issue. Unfortunately we only support the AppImage.

Please raise this issue with the maintainer of the AUR repository https://aur.archlinux.org/packages/cura-git/

closed 2020-11-11 by jellespijker