Ultimaker/Cura #8732: Cura 4.8.0 crashes during first-time-setup immediately after installation

JAL741, 2020-11-11:
I installed Cura 4.8.0.  It uninstalled my working Cura 4.7 (during which I chose the options to also remove all profiles) and then installed itself.  It then asked me the first-time-setup questions about license agreement and data collection.   When it asked me to sign it with an account; I pressed 'skip'.  Then it asked me to select my printer; I pressed "non-networked printer".  Before the list of printers to choose from was displayed, the application window disappeared completely without warning.  Looking in Windows Event Logs, I see the following failure logged:

Faulting application name: Cura.exe, version: 0.0.0.0, time stamp: 0x5f9933ba
Faulting module name: ig4icd64.dll, version: 9.17.10.4459, time stamp: 0x573dfb41
Exception code: 0xc0000005
Fault offset: 0x00000000000011b1
Faulting process id: 0x1a68
Faulting application start time: 0x01d6b84cc744f216
Faulting application path: C:\Program Files\Ultimaker Cura 4.8.0\Cura.exe
Faulting module path: C:\WINDOWS\system32\ig4icd64.dll
Report Id: 575bc833-21a5-4c85-bfdf-5bb239c17d86
Faulting package full name: 
Faulting package-relative application ID:

JAL741, 2020-11-11:
I have now un-installed Cura 4.8.0, then re-installed it.  The crash now happens even before being asked to select a printer.

I have captured a video demonstrating this (attached).  I have also attached the entire contents of the %AppData%\Cura folder where I see several log files that developers may find useful.

[Cura 4.8.0 installaiton failure (crash) - 11-Nov-2020.zip](https://github.com/Ultimaker/Cura/files/5525840/Cura.4.8.0.installaiton.failure.crash.-.11-Nov-2020.zip)
[Cura AppData-Roaming files.zip](https://github.com/Ultimaker/Cura/files/5525850/Cura.AppData-Roaming.files.zip)

.

jellespijker, 2020-11-11:
Hi @JAL741 

Could it be that I asked you on a Facebook comment to report this here? If so. Thank you for your the bug reports and log files. These will be most helpful, but you would make us really happy if you also fill in the bug template next time (lets hope there will be no next time tho). Since that template also contains a question regarding the graphic driver and operating system that you're using.

I haven't look in detail through the logs but the faulting module that is reported by windows is C:\WINDOWS\system32\ig4icd64.dll, which is a part of the Intel Graphics Accelerator Drivers for windows 7.

Can you tell us what videocard  and which windows system you are using?

CGBromley, 2020-11-11:
I am having same problem but I am using Windows 10, Intel(R) HD Graphics.  Can I just go back to 4.7?  I don't see where I can download, keeps sending me to 4.8.0

JAL741, 2020-11-11:
@jellespijker system information is now attached:

[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5526790/DxDiag.txt)
[SysInfo.txt](https://github.com/Ultimaker/Cura/files/5526793/SysInfo.txt)

JAL741, 2020-11-11:
@jellespijker your suggest that it might be related to the graphics processor in my laptop, reminded me that there are two; the integrated Intel CPU graphics processor, and an add-in nVidia processor.  By defualt, the laptop automatically selects which graphics processor to use, per application.  I decided to force Cura.exe to use the nVida graphics processor, and that allowed me to complete then Cura 4.8.0 installation and setup.  It is now workign for me, but the fact that Cura crashed in the first place, instead of handling this gracefully, is still a bug in my opinion.

[image]

[image]

jellespijker, 2020-11-16:
@CGBromley Older versions of Cura can be downloaded from https://github.com/Ultimaker/Cura/releases 
You can have multiple versions installed along side each other

jellespijker, 2020-11-16:
@JAL741 Glad to hear that you have found a work around, and yes you're absolutely correct that this is still a bug. I will ask the developers whom work on Windows to take a closer look, since they're more familiar with this system then I'm. 

@konskarm or @rburema could either one of you take a look at this issue please?

@JAL741 pressing the "new issue" button should have presented this screen
[image]

rburema, 2020-11-17:
Looking through the logs, I see that it somehow detects OpenGL 3.1 with GLSL version 140, and it's running from your Intel card at the time the logs stop:

```
2020-11-11 10:47:20,170 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [210]: Support for Vertex Array Objects: True
2020-11-11 10:47:20,188 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2020-11-11 10:47:20,196 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  3.1.0 - Build 9.17.10.4459
2020-11-11 10:47:20,203 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   Intel
2020-11-11 10:47:20,210 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: Intel(R) HD Graphics 3000
2020-11-11 10:47:20,218 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    1.40.0
```

So there are 2 issues here:
- Why isn't is selecting your NVidia card when you select that (we can't do much about this)
- Why is it crashing rather than (or after!) selecting the legacy rendering option that should take care of all OpenGL drivers supporting less than 4.1.x? (And a hard crash at that, not even an error report shown.) We've had very hard to debug problems with Intel drivers before (frustratingly, on their side, so we can do little but work around them).

I'm not sue what's wrong, but you could try these things:
- Update the Intel drivers (I'm not sure how recent yours' are).
- Make sure the program _does_ start with the NVidia card.
- Set `force_empty_shader_cache` under the `[view]` heading to `= True` in `cura.cfg`.

jellespijker, 2020-11-18:
> Why isn't is selecting your NVidia card when you select that (we can't do much about this).

@rburema The log you're reading was when he was indeed running from the Intel card. Selecting the NVidia card was a work around and then it doesn't crash (therefor no log file provided of that scenario)

md-hexdrive, 2020-11-25:
I also have this identical problem, and cura 4.8 also crashes for me at first setup. I haven't seen the log file itself, but my computer only has Intel Integrated Graphics.
I will be moving to an earlier version of cura for now.

fabriciofx, 2020-12-16:
@jellespijker I also have the same problem. My setup is Windows 10 20H2, 8GiB RAM, CPU intel and GPU Intel Integrated Graphics (It's a ProBook 6470b HP notebook).

ondrapCZE, 2020-12-27:
Hi, I am not sure if my laptop have the same issue but manifestation is same. During the first time setup it crashes. Attached is the cura error output. [
[stderr.log](https://github.com/Ultimaker/Cura/files/5744556/stderr.log)
](url)

fabriciofx, 2020-12-27:
I haven't sent my logs, but there aren't errors on it, just few WARNING, INFO and DEBUG information:
```
2020-12-23 22:01:57,751 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ConsoleLogger
2020-12-23 22:01:57,781 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraDrive
2020-12-23 22:01:57,816 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [92]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.8.0\CuraEngine.exe
2020-12-23 22:01:57,823 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraEngineBackend
2020-12-23 22:01:57,840 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileReader
2020-12-23 22:01:57,844 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2020-12-23 22:01:57,853 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura 4.8.0\CuraEngine.exe
2020-12-23 22:01:57,856 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.8.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2020-12-23 22:01:57,869 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileWriter
2020-12-23 22:01:57,883 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJReader
2020-12-23 22:01:57,897 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJWriter
2020-12-23 22:01:57,917 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLReader
2020-12-23 22:01:57,932 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLWriter
2020-12-23 22:01:57,949 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-12-23 22:01:57,977 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-12-23 22:01:57,992 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-12-23 22:01:58,005 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-12-23 22:01:58,019 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-12-23 22:01:58,033 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-12-23 22:01:58,068 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2020-12-23 22:01:58,092 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2020-12-23 22:01:58,122 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2020-12-23 22:01:58,130 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2020-12-23 22:01:58,135 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.8.0
2020-12-23 22:01:58,140 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2020-12-23 22:01:58,145 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2020-12-23 22:01:58,151 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2020-12-23 22:01:58,156 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-12-23 22:01:58,160 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-12-23 22:01:58,164 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2020-12-23 22:01:58,168 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2020-12-23 22:01:58,172 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2020-12-23 22:01:58,176 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-12-23 22:01:58,181 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-12-23 22:01:58,185 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2020-12-23 22:01:58,190 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]
2020-12-23 22:01:58,194 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-12-23 22:01:58,198 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-12-23 22:01:58,203 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-12-23 22:01:58,207 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-12-23 22:01:58,211 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2020-12-23 22:01:58,227 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2020-12-23 22:01:58,246 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2020-12-23 22:01:58,261 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
[... cut ...]
2020-12-23 22:02:09,166 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray_composite.shader]...
2020-12-23 22:02:09,192 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\platform.shader]...
2020-12-23 22:02:09,213 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\xray.shader]...
```

Ghostkeeper, 2020-12-28:
Over in https://github.com/Ultimaker/Cura/issues/9018 they've found a possible fix/workaround for this issue: Reinstall Visual C++ Redistributable, or uninstall the ASUS GPU Tweaks. Something there seems to be messing with your graphics drivers that causes a crash the first time it needs to clip something (to display a scrollable view).

dgiambo1, 2020-12-28:
I thought the work-around was good. Before installing vcredist it wouldn't get past the data collection screen, after doing this it let me past that screen but it only got me to this issue. I clicked "Add Non-Networked Printer" and it crashed again. Same crash each time. Not sure what is going on, but I can't use Cura anymore, 4.7.1 and 4.8.0 crash in the same way. For me, everything was working fine until I tried to add a new printer, which somehow ended up clearing my configuration and showing me all the 'first-time user' stuff again, then I ended up here...

dgiambo1, 2020-12-28:
Even more confused. I let the machine sit awhile. Just tried again so I could collect a log. It didn't crash on data collection or on the add non-networked printer this time. That's after dozens of failures earlier today. Could it be trying to reach a server or something during this phase? Something outside of my PC seems to have changed the behavior.

fabriciofx, 2020-12-28:
@Ghostkeeper how can I reinstall *only* the Visual C++ Redistributable?

P.S. I'm using Windows 10 20H2 and my notebook has an Intel HD 4000 with all intel drivers installed. I also have an external monitor, which is the main monitor in windows.

nordlead2005, 2020-12-29:
FYI, this has nothing to do with ASUS GPU Tweak, I just listed it as a program that needs newer VC++ redistributables, while Cura seems to be crashing on them even though they should work. VC++ combined all the redistributables from 2015-2019, where as previously you would have installed, 2015, 2017 & 2019 all as separate versions.

I also recently discovered that MeshMixer doesn't work without the 2015-2019 Redistributable, so rolling back gets cura working, but breaks other essential 3d printing tools. I'm fairly close to abandoning Cura over this.

I also don't know if this is actually a duplicate, I haven't looked at the logs in this issue, but it is fairly simple to roll back VC++ Redistributables and test if it works for you. 

@fabriciofx 

1. Open Add or Remove Programs
2. Search for redistributable
3. uninstall the 2015-2019 x64 redistributable (leave the 32-bit as it doesn't matter, because Cura is installed as 64-bit)
4. In a file explorer, navigate to your cura install directory (typically C:\Program Files\Ultimaker Cura 4.8.0)
5. scroll to the bottom and execute vcredist_x64.exe (this is 2015 ONLY, so anything that needs 2017 or 2019 will no longer function properly)
6. Reboot PC

Now, it is possible that there is a 2015-2019 redistributable version that does work. However, it is hard to find anything other than the latest to download. I tried 1 version back, but I don't believe that worked. I'm not even sure when I updated the redistributable or how big my jump was since Cura 4.8.0 was working a week before Christmas for me, and then broke, but that doesn't exactly line up with MS's releases.

Hoghunterx, 2020-12-30:
Hello all I had to join this forum to help everyone out. I figured out how to install Cura 4.8 on Windows 10. You need to shut off everything that is under Windows firewall. Click on the tab that says app and browser control. Scroll down to exploit protection. Click on exploit protection settings. Shut all of those settings off by default. Restart computer reinstall software it should run. It works fine for me. Hope this helps please let me know if it worked for you

fabriciofx, 2020-12-30:
@Hoghunterx thanks!!! It worked for me too!!! But there is a problem: to run Cura 4.8 I need to keep all exploit protection disabled... So, I don't know if it is a problem with Cura or Windows. @nordlead2005 @Ghostkeeper @jellespijker Any idea?

Hoghunterx, 2020-12-30:
You have to keep the exploits off at all times or it will not work I'm glad
I helped you out brother

Hoghunterx, 2020-12-30:
The computer I use for my 3D printer I don't put on the internet so I don't
have to worry about any malicious attacks

Ghostkeeper, 2020-12-30:
> @Ghostkeeper how can I reinstall only the Visual C++ Redistributable?

The redistributable should be somewhere in your installation folder. If not, you can also download the one that ships with Cura's installer here: https://github.com/Ultimaker/cura-binary-data/tree/master/windows

Ghostkeeper, 2020-12-30:
I don't know what this exploit protection is, but if it's part of the Windows firewall I shouldn't recommend turning it off. Cura is open source and you can see exactly what it's doing; albeit you'd have to trust that you've downloaded the program generated from that source code or generate CuraEngine+libArcus yourself.

However Cura is calling here upon `ig4icd64.dll`, as indicated by the crash report of JAL741. This is your Intel graphics driver. I don't suppose there would be an exploit in the Intel graphics driver, but graphics drivers do strange things sometimes so I can imagine that it could trigger some flags.

Hoghunterx, 2020-12-30:
When I download The Cura file from the website, the Windows defender
insignia is on the file. The program would only install halfway and act
like it was done. When it would try to run the software it would start up,
then when you get to the point of adding a printer it would immediately
crash.
   I tried to install the program on another computer of mine, it worked no
problem. Why it worked fine on this computer I don't know, but on that
installation when it got halfway it asked me to allow permission I'm
assuming from two other external companies that are probably
part-developing that software.
   I read on another forum about earlier versions of Cura, to shut off the
Windows firewall during installation.
    I just got lucky with finding the exploit protection tab and shutting
everything off and it worked. I don't recommend running without a firewall
but I don't use my computer online. I ran Cura all night last night not one
problem. I did not update my graphics card everything works fine.
   I just turned my Windows firewall back on and all exploit protection
back on my software is running fine. I am not a computer guy just trying to
pass on information

AceManJr, 2020-12-30:
Had all the similar problems as stated above. The one thing that succeeded in successfully installing CURA was turning off the private windows firewall and turning on after install. 
[image]

Thanks for the suggestion!

fabriciofx, 2020-12-31:
@Hoghunterx I appreciate your information a lot! Thanks, again! 😄 
@Ghostkeeper I don't know if is something around the video drivers. I *never* have had a problem like it and use many 3D applications on my actual setup. Maybe Cura is trying accessin internet and Windows doesn't allow. Can be it?

doc-Seltsam, 2020-12-31:
#8367

fabriciofx, 2021-01-02:
@Hoghunterx @Ghostkeeper @doc-Seltsam I think discovered root of the cause: the Windows Address Space Layout Randomization (ASLR). I've disabled it just in my Windows 10 and Cura worked like a charm! Please, test it!

Ghostkeeper, 2021-01-06:
> When I download The Cura file from the website, the Windows defender insignia is on the file.

Indeed, that's because the installer requires administration access to be able to install anything into Program Files.

> Had all the similar problems as stated above. The one thing that succeeded in successfully installing CURA was turning off the private windows firewall and turning on after install.

Wait, so after the installer has finished you turn the firewall back on and Cura runs fine? That would mean that the firewall is blocking something from the *installer*, rather than blocking something from Cura? And the installer perhaps didn't install everything?
The Cura installer is a fairly standard [NSIS](https://en.wikipedia.org/wiki/Nullsoft_Scriptable_Install_System) installer which is used by loads of software. Perhaps the firewall would block the installation of the C++ redistributable? But it's the same redistributable that we've been shipping with Cura for 5 years. I can't imagine that the Windows firewall would block a common Microsoft product like the C++ redistributable either.

> Maybe Cura is trying accessin internet and Windows doesn't allow. Can be it?

I'm fairly certain that this is not it since the crash report posted here indicates that the Intel graphics drivers crashed, which has nothing to do with the network or the internet. Cura accesses the network to find printers on the Wifi network, which would trigger Windows Defender to ask the user if Cura should be allowed access. Denying this access has no repercussions aside from that Cura doesn't get any response from querying the network for printers and then there will be no printers on the network.

> I think discovered root of the cause: the Windows Address Space Layout Randomization (ASLR). I've disabled it just in my Windows 10 and Cura worked like a charm! Please, test it!

I use ASLR in my Windows installation and it's not crashing, so I can't test this. I don't have Intel hardware though so it makes sense that it's not crashing for me. But it's not just ASLR.
ASLR is quite an important security feature so maybe disabling it is not the best idea.

TheOutcast83, 2021-02-12:
I just tried to run Cura 4.8 on my Windows7/10 laptop using the "Run as Administrator" function and it let me add my one non-networked printer that way with out crashing.

ctmal956, 2021-03-11:
I'm new at Cura(just downloaded it tonight to check it out) and I was having this issue.  I have an Intel 3000 graphics card, I'm not sure if this is the issue.
I seemed to have resolved it by right clicking the cure.exe file, select Troubleshoot compatibiliy and let it check the compatability.  It should set it to Windows 8.  After that, the program is opening up fine for me.

ukprinter, 2021-03-24:
Not sure if my problem is related BUT I have several instals of cura up to 4.8 which I was happy with until today.  I tried to run it on a Windows 10 64 bit machine that it was installed on months and was working fine till recently.

When I tried to run it windows seemed to think it was the first time I have run 4.8 and I seem to have lost my machines and configurations.

I went back to 4.7 and that is fine.   Its only 4.8 which seems weird to me.

GaryStebbins, 2021-03-25:
Cura 4.8.0 on Windows 10 as upgraded from previous versions was working for me. My hard drive died, and I did a full system rebuild from scratch. The freshly installed Cura 4.8.0 crashed whenever I tried to add a printer. I then ran Cura "as administrator" and was able to install the printer, and can now run Cura normally without problem.

fabriciofx, 2021-04-02:
Guys, I've some good news. I've updated my video drivers (my video card board is a Intel HD Graphics 4000 and I was using the Windows 10 default drivers) and now Cura 4.8 is working as a charm, without dissable any Windows 10 protection or needs to running as administrator! Thanks for all!

rhubiere, 2021-05-30:
Not sure this will help anyone but I came across this post today.  For what its worth, nothing has helped my case other than continuing to rely on v4.6. [https://support.ultimaker.com/hc/en-us/articles/360018633500-Ultimaker-Cura-won-t-run-on-my-Ryzen-Graphics-card](url)

Hoghunterx, 2021-05-31:
Windows security is the only thing that is screwing with this software if
you run as an administrator it will work. I run this software every day I
love that everybody keeps replying to this string I hated it when I
couldn't get it going. If you don't run as an administrator you have to
shut every Windows security option off then install it will work. I'd be
more than happy to post a YouTube video if y'all would like it and I'll
show you exactly how I do it thanks for all the replies keep it going

Hoghunterx, 2021-05-31:
That's why I don't put my laptop online I don't do windows 10 updates I got
the first version of Windows 10 and I could run everything fine

GregValiant, 2024-11-20:
Is this still an issue in current versions of Cura (5.8.0 and up)?  Can this be closed?

fabriciofx, 2024-11-20:
@GregValiant I think so. I'll close it.

closed 2024-11-20 by GregValiant