Ultimaker/Cura #8733: starting 4.8 encountering error with profile

g097103, 2020-11-11:
[image]

This error does not occur in 4.6.1 using the same profiles. For obvious reasons, I do not allow it to delete existing profiles which work well in 4.6.1

nallath, 2020-11-12:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

Could you also provide your configuration folder?

g097103, 2020-11-12:
I delete the m200 profile which was a monoprice mini, uninstalled 4.8 the reinstalled 4.8 and it seems to have resolved this issue.

closed 2020-11-12 by g097103