Ultimaker/Cura #8734: Attempting to save project fails

g097103, 2020-11-11:
Trying to save 3mf file produces the following and an inability to open the 3mf file afterward.

**Traceback (most recent call last):**

  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 1063, in event
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 464, in event
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Event.py", line 218, in call
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\Utils\Threading.py", line 34, in _handle_call
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\3MFWriter\ThreeMFWorkspaceWriter.py", line 57, in write
    self._writeContainerToArchive(container, archive)
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\3MFWriter\ThreeMFWorkspaceWriter.py", line 154, in _writeContainerToArchive
    serialized_data = container.serialize(ignored_metadata_keys = ignore_keys)
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 269, in serialize
    for instance in container.findInstances():
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Settings\InstanceContainer.py", line 649, in findInstances
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Settings\InstanceContainer.py", line 637, in _instantiateCachedValues
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Settings\InstanceContainer.py", line 733, in getDefinition
UM.Settings.InstanceContainer.DefinitionNotFoundError: Could not find definition creality_cr10s4 required for instance generic_petg_175_creality_cr10s4

**Logs**

Thread 0x00002984 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000013cc (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000029a0 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001b24 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001f4c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000444 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000062c (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002eec (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002ca8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002bb0 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002ca0 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 549 in wait
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\Utils\Threading.py", line 39 in _call_on_qt_thread_wrapper
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\FileHandler\WriteFileJob.py", line 76 in run
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 132 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001b10 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002a40 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000029d8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000200c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002580 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00002ffc (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.8-exe\build\inst\bin\cura_app.py", line 167 in exceptHook
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 847 in run
  File "X:\4.8-exe\build\inst\bin\cura_app.py", line 235 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

SilverThunderbird, 2020-11-12:
Installed 4.8, opened a model. Sliced. Tried to Save Project. Get an error every time. Tried uninstalling/reinstalling.  Same issue.
Log and screen shot attached.

[cura.log](https://github.com/Ultimaker/Cura/files/5531012/cura.log)
[image]

konskarm, 2020-11-12:
Please fill in the entire bug template https://github.com/Ultimaker/Cura/issues/new?assignees=&labels=Type%3A+Bug&template=bug-report.md&title=. Which Cura version does it occur? I cannot reproduce it on 4.8. Steps I followed:

- Added a Creality CR10-S4 
- Added a model
- Sliced
- Saved project

My .3mf file was produced without an issue. So I am guessing there are more reproduction steps that are missing. Does it work fine for 4.7? If it does, can you create a project file in 4.7 that causes the problem to occur in 4.8 and post it here?

g097103, 2020-11-12:
> Please fill in the entire bug template https://github.com/Ultimaker/Cura/issues/new?assignees= <https://github.com/Ultimaker/Cura/issues/new?assignees=&labels=Type%3A+Bug&template=bug-report.md&title=> &labels=Type%3A+Bug&template=bug-report.md&title=. Which Cura version does it occur? I cannot reproduce it on 4.8. Steps I followed:

Will update bug report.

Does not happen in 4.7, I save 3mf files there all the time.

I have a cr10, not a CR10-S4 which is how it is in the profile. Not sure where the S4 is coming from.

Was able to open a 4.7 3mf file.

 

Scott Morris

(813) 810-6914

Tampa Power Squadron SEO -  <https://tampaps.org/> https://tampaps.org             <https://myfloatplan.org/> https://myfloatplan.org 

Stf/C Boat Systems

g097103, 2020-11-12:
[DualEarSaverScaled.zip](https://github.com/Ultimaker/Cura/files/5531524/DualEarSaverScaled.zip)

konskarm, 2020-11-12:
When trying to load that project in Cura I get the following:
```
2020-11-12 16:35:47,686 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/k.karmas/Downloads/DualEarSaverScaled/DualEarSaverScaled.3mf
2020-11-12 16:35:47,702 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]: Exception: Could not check file PyQt5.QtCore.QUrl('file:///C:/Users/k.karmas/Downloads/DualEarSaverScaled/DualEarSaverScaled.3mf')
2020-11-12 16:35:47,714 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]: Traceback (most recent call last):
2020-11-12 16:35:47,742 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:   File "C:\dev\Cura\cura\CuraApplication.py", line 1922, in checkIsValidProjectFile
2020-11-12 16:35:47,752 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:     result = workspace_reader.preRead(file_path, show_dialog=False)
2020-11-12 16:35:47,765 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:   File "C:\dev\Cura\plugins\3MFReader\ThreeMFWorkspaceReader.py", line 219, in preRead
2020-11-12 16:35:47,774 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:     archive = zipfile.ZipFile(file_name, "r")
2020-11-12 16:35:47,793 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:   File "C:\Python\Python37\lib\zipfile.py", line 1225, in __init__
2020-11-12 16:35:47,802 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:     self._RealGetContents()
2020-11-12 16:35:47,812 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:   File "C:\Python\Python37\lib\zipfile.py", line 1292, in _RealGetContents
2020-11-12 16:35:47,841 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]:     raise BadZipFile("File is not a zip file")
2020-11-12 16:35:47,852 - ERROR - [MainThread] cura.CuraApplication.checkIsValidProjectFile [1925]: zipfile.BadZipFile: File is not a zip file
```

Even when trying to open it with winrar (since 3mf files are basically zip files) I get an error:
[image]

Also I cannot even open it in 4.7.1. Same result. Something is going wrong elsewhere. Do you mind zipping your entire configuration folder and sharing it here? You can find the config folder by going to Help -> Show Configuration Folder in Cura.

g097103, 2020-11-12:
my config folder is zipping up as 25M so I can't upload it here although most of the folders show up as empty.

konskarm, 2020-11-12:
Could you upload it in https://wetransfer.com/ and post a link here?

g097103, 2020-11-12:
https://we.tl/t-mxkKnHVaIq

konskarm, 2020-11-13:
Indeed it is crashing when saving a project file. After further inspection, it seems to be caused by the Startup Optimizer plugin, because when I do "Extensions -> Startup Optimizer -> Restore all configuration files" and restart Cura, the problem no longer occurs. We do not maintain this plugin, so please post this issue in the [github page of the Startup Optimizer](https://github.com/fieldOfView/Cura-StartOptimiser).

closed 2020-11-13 by konskarm

SilverThunderbird, 2020-11-13:
Third party plugins must be a thorn in every developer's side.
Thank you _so much_ for the speedy resolution to this issue! 
Have a great day, and stay safe.

g097103, 2020-11-13:
Thanks for the quick resolution. I’ll follow up with the developer.

 

Scott Morris

(813) 810-6914

Tampa Power Squadron SEO -  <https://tampaps.org/> https://tampaps.org             <https://myfloatplan.org/> https://myfloatplan.org 

Stf/C Boat Systems

fieldOfView, 2020-11-13:
> Third party plugins must be a thorn in every developer's side.

Gee, thanks.

g097103, 2020-11-13:
> Links:

Wasn't me that said that. I see you found the original report, I thank 
you for your quick response to this.

---
Scott Morris
SEO Americas Boating Club of Tampa Bay
Stf/C Boat Systems
813-810-6914

Evil Overlord Rule 7 - No matter how desperate I become I will never 
consume any energy field larger than my head.

nallath, 2020-11-13:
> > Third party plugins must be a thorn in every developer's side.
> 
> Gee, thanks.

@fieldOfView 
[image]

konskarm, 2020-11-13:
> Third party plugins must be a thorn in every developer's side.

I missed that. No. On the contrary they are extremely useful to our users and they are contributed by people who build and maintain them in their free time without asking anything back.

It's good to appreciate the hard work that goes into that.

SilverThunderbird, 2020-11-13:
> 
> 
> > Third party plugins must be a thorn in every developer's side.
> 
> Gee, thanks.

That was me.  Certainly not a slight to plugin developers, and disappointed if it was taken that way.

I _love_ my plugins, but have to sympathize with parent app developers when a change breaks a plugin, or the once compliant plugin breaks the app (for whatever reason) and causes the parent to abend. The first place people look to blame is the parent developers.

Ghostkeeper, 2020-11-17:
The bugs in plug-ins are a small hindrance compared to the benefit they bring. Quite often it saves us development time too, because there is much less priority to implement a feature that is already implemented through some plug-in.

SilverThunderbird, 2020-11-17:
Absolutely. Plugins can make a great app a fantastic app, and some are
virtually indispensable.