Ultimaker/Cura #8736: CURA 4.8 Layer Height Slicing Issue

Ed-EE-Eng, 2020-11-11:
**Application version**
Cura 4.8

**Platform**
Windows 10

**Printer**
Custom - Tronxy X5S - SKR 1.4T/Marlin 2.0.7.2

**Reproduction steps**
1. Loaded Bench.stl file into Cura
2. Used my standard setup for PLA, Layer Height: 0.2 mm, Initial Layer Height: 0.2 mm
3. Ran Slicer, and ended up with extraordinarily long slicing time, very long estimated print time of 7hrs 35Min, and very high filament usage: 50.45m.
4. Tried different Layer Heights and re-ran slicer. Example attached was layer Height: 0.2 mm, Initial Layer Height 0.21 mm.  Now, slicing time was quick, estimated print time as expected (<1 Hr), filament usage: 3.6m as expected.  Ran this file on my printer (through Octoprint) and print came out "perfect".

5. Went back to .2/.2 mm layer Ht and same long print time resulted. Problem is repeatable when ever using .2/.2 combination.  All others I tried (.1/.1, .15.15, .2/.3, etc) all produced expected (normal) slicing and time/material estimates.
6 I then loaded other stl files and the problem did not manifest.

**Screenshot(s)**
See images.

**Actual results**
See discussion points above

**Expected results**
See discussion points above

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
...ZIP file is 3MF project file for this bug report.

[PLA_Cura4.8_LH_.2_InitLLH.zip](https://github.com/Ultimaker/Cura/files/5526965/PLA_Cura4.8_LH_.2_InitLLH.zip)
[image]
[image]
[image]

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-11-12:
Hi @Ed-EE-Eng. It's good to hear that the zits are no longer there!
Regarding the issue, this quite interesting! When I try to do it with other printers, it does not occur, but it reproduces with the project file. And not just that, it also generates an infinite outer wall! This makes me think that a division by 0 may be occuring somewhere.

[image]

I'll put it up for discussion with the team.

Ghostkeeper, 2020-11-17:
This was put in our backlog. Developers, refer to CURA-7846.

rmeissn, 2020-11-26:
I get printed zits with Cura 4.8 (like on your last picture of the initial post for Cura 4.7.1) wherever a line starts/stops in the outer shell. Depending on the "Z Seam Alignment" setting they are all aligned or randomly placed all over the print. Haven't tried Cura 4.7.1 for comparison yet.

Treeed, 2021-05-12:
I also get the infinite wall issue with the "fine" profile when I set the horizontal expansion to -0.05 (find a project file with the specific model that triggers it attached). When I even slightly alter the expansion the issue doesn't occur anymore. 
Trying to print the file the printer just stops at the problematic layer and tells me "coordinates go out of the print area" or something the like.
The model was checked with Netfabb Premium and doesn't show any signs of defects (including self intersections).

[image]
Looks like I'm trying to print a laser cannon

[UM2_slider_bug.zip](https://github.com/Ultimaker/Cura/files/6466914/UM2_slider_bug.zip)

Ghostkeeper, 2021-05-20:
I must say that I can't reproduce the issue any more. It's still in our backlog however.

Treeed, 2021-05-20:
Have you tried my .3mf file? For me it is reproducible with Cura 4.9.1. To reproduce from scratch:
- set the printer to UM2+
- set the nozzle to 0.25
- set the profile to fine
- set the horizontal expansion to -0.05
- import the stl from [infinite wall bug.zip](https://github.com/Ultimaker/Cura/files/6516611/infinite.wall.bug.zip)
- slice and get a laser with a line to infinity
- ???
- profit

Ghostkeeper, 2021-06-03:
Yes, I used your project file, but it doesn't reproduce the issue for me on Master nor on 4.9.1:
[image]

In your alternative reproduction steps I couldn't set the profile to "fine" for step 3, because the 0.25mm nozzle only has an "extra fine" profile with PLA on the UM2+.
The result is more or less the same though, although this time the model is centred in the build plate:
[image]

Since I'm not able to reproduce the issue in 4.9.1, but you are, the only conclusions can be that:
* It could depend on the Cura configuration. This seems unlikely because we have the same settings and profiles (as per the project file) and the only really relevant preferences here are the colour scheme and compatibility mode, which we have set to the same preferences.
* It could be random. You can never know.
* It could depend on the system environment. Maybe it's some bug in your graphics driver. Maybe it's something that we left uninitialised and different graphics drivers handle the default differently. Maybe it's something dependent on the operating system.

The most likely culprit would be the graphics driver, I think.

Treeed, 2021-06-05:
Interesting that you still can't reproduce the issue. You are right that it should be the "Extra Fine" profile. 

- To rule out issues with the configuration I have completely removed my installation folders %AppData%\cura, %LocalAppData%/cura and %ProgramFiles%\Ultimaker Cura 4.9.1, then downloaded a fresh installer, installed it and set it up with the aforementioned steps. It still happens.
- It happens for me every time, no matter where i place the model 
- I'm not sure of the slicer uses any graphics resources? The bug is not visual, the infinite line is actually there, print time goes up to 15h and when trying to print the part the printer will stop at the relevant layer with a message along the lines of "the move goes out of the build volume".
The issue is also reproducible on my laptop. These are the two environments FWIW:

Desktop:
Windows 10 Enterprise
i5-3570K
Nvidia GTX 970

Laptop:
Windows 7 Ultimate
i5-M460
ATI Mobility Radeon HD 5650

[image]

Ghostkeeper, 2021-06-14:
> I'm not sure if the slicer uses any graphics resources?

No, it doesn't. I was thinking that it could be a visualisation issue. However since you say that it's actually in the g-code, that is ruled out I think.

It's very weird. CuraEngine has hardly any platform-specific code. The only thing I can think of is the disabling of multi-threading for MacOS.

fvrmr, 2022-02-01:
The linked ticket on our backlog has been closed because we can't reproduce this anymore in Arachne beta 2. So I will close this issue.