Ultimaker/Cura #8737: Print settings menus shown incorrectly (symbols < and ∨ in menus)

hummel65, 2020-11-11:
**Application version**
4.8.0

**Platform**
macOS 10.15.7 and Windows 10 OS Build 19041.630

**Printer**
Ultimaker 3 and Ultimaker 2+

**Reproduction steps**
1. Start Cura
2. Open Print settings
3. Click any of the settings like "Quality", "Speed"
4. < is not changing to V

"Quality" and all others except "Dual Extrusion" is always <
"Dual Extrusion" is always V

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

[image]
[image]
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
When content of "Quality" is shown < is expected to change to V and so on for all items

**Log file**
2020-11-11 23:57:20,587 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-11-11 23:57:20,591 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-11-11 23:57:20,596 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin AMFReader
2020-11-11 23:57:20,600 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleView
2020-11-11 23:57:20,605 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-11-11 23:57:20,611 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-11-11 23:57:20,615 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-11-11 23:57:20,623 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-11-11 23:57:20,628 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-11-11 23:57:20,648 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MaterialCostTools
2020-11-11 23:57:20,666 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [110]: Device 3Dconnexion / Space Explorer, profile spacemouse
2020-11-11 23:57:20,667 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [127]: axis 0, scale = 0.100000, threshold = 0.010000, offset = 0.000000, target = movx
2020-11-11 23:57:20,667 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [127]: axis 1, scale = 100.000000, threshold = 1.000000, offset = 0.000000, target = zoom
2020-11-11 23:57:20,667 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [127]: axis 2, scale = 0.100000, threshold = 0.010000, offset = 0.000000, target = movy
2020-11-11 23:57:20,667 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [127]: axis 3, scale = 0.050000, threshold = 0.010000, offset = 0.000000, target = rotpitch
2020-11-11 23:57:20,667 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [127]: axis 4, scale = 0.050000, threshold = 0.010000, offset = 0.000000, target = rotroll
2020-11-11 23:57:20,667 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [127]: axis 5, scale = -0.100000, threshold = 0.010000, offset = 0.000000, target = rotyaw
2020-11-11 23:57:20,668 - DEBUG - [RawMouse] RawMouse.RawMouse._run [167]: Trying to open IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS10@14800000/4-Port USB 2.0 Hub@14800000/AppleUSB20Hub@14800000/AppleUSB20HubPort@14820000/4-Port USB 2.0 Hub@14820000/AppleUSB20Hub@14820000/AppleUSB20HubPort@14821000/4-Port USB 2.0 Hub@14821000/AppleUSB20Hub@14821000/AppleUSB20HubPort@14821300/Space Explorer@14821300/IOUSBHostInterface@0/ConnexionUSB@14821300,0
2020-11-11 23:57:20,670 - INFO - [RawMouse] RawMouse.RawMouse._run [173]: Manufacturer: 3Dconnexion
2020-11-11 23:57:20,673 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RawMouse
2020-11-11 23:57:20,673 - INFO - [RawMouse] RawMouse.RawMouse._run [174]: Product: Space Explorer
2020-11-11 23:57:20,680 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PauseBackendPlugin
2020-11-11 23:57:20,686 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CustomJobPrefix
2020-11-11 23:57:20,692 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OrientationPlugin
2020-11-11 23:57:20,696 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GcodeFilenameFormat
2020-11-11 23:57:20,700 - INFO - [MainThread] StartOptimiser.StartOptimiser.__init__ [40]: 1919 containers are blacklisted by StartOptimiser
2020-11-11 23:57:20,705 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin StartOptimiser
2020-11-11 23:57:20,953 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SettingsGuide2
2020-11-11 23:57:20,960 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ZOffsetPlugin
2020-11-11 23:57:20,965 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CustomSupports
2020-11-11 23:57:20,967 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [432]: Loading all plugins took 1.0972840785980225 seconds
2020-11-11 23:57:20,975 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2020-11-11 23:57:20,994 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-11-11 23:57:20,998 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-11-11 23:57:21,001 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-11-11 23:57:21,004 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-11-11 23:57:21,007 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-11-11 23:57:21,010 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
[... 85 lines cut ...]
2020-11-11 23:57:28,397 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
2020-11-11 23:57:28,404 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/striped.shader]...
2020-11-11 23:57:28,411 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/xray.shader]...
2020-11-11 23:57:28,417 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/xray_composite.shader]...
2020-11-11 23:57:28,424 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/default.shader]...
2020-11-11 23:57:28,430 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/grid.shader]...
2020-11-11 23:57:28,437 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/platform.shader]...
2020-11-11 23:57:28,486 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.0.app/Contents/MacOS/resources/shaders/xray.shader]...
2020-11-11 23:57:28,695 - INFO - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDevice.connect [114]: Attempting to connect to cluster 4QNMc9v-xHiOGMg-P_3CuMe6CYIorv0C7yD83CF3nrlB
2020-11-11 23:57:28,737 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2020-11-11 23:57:28,747 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2020-11-11 23:57:30,541 - DEBUG - [Thread-5] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for ultimaker2_plus took 1.6376550197601318 seconds.
2020-11-11 23:57:30,545 - DEBUG - [Thread-5] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2020-11-11 23:57:31,450 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 5.0s
2020-11-11 23:57:32,402 - INFO - [Thread-4] SettingsGuide2.CuraSettingsGuide.load_all [161]: Finished loading Settings Guide articles.
2020-11-11 23:57:48,519 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
[... cut ...]
2020-11-11 23:57:55,378 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2020-11-11 23:57:55,631 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2020-11-11 23:57:55,631 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [321]: Killing engine process
2020-11-11 23:57:55,632 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [324]: Engine process is killed. Received return code -15

**Additional information**
This is n minor but annoying UI bug.

konskarm, 2020-11-12:
I am closing this as a duplicate of https://github.com/Ultimaker/Cura/issues/8283. Please add any extra comments there.

closed 2020-11-12 by konskarm

hummel65, 2020-11-12:
Hi,

I searched for it and couldn’t find it, sorry for my duplicate report. Thank you for the speedy reply!

Mats