Ultimaker/Cura #8739: [4.8.0b] Tree support inconsistency

VicOnTheBeach, 2020-11-12:
**Application version**
4.8.0b

**Platform**
Windows 10 Familly x64

**Printer**
CREALITY CR10s Pro

**Reproduction steps**
0. Download Cura project here https://www.thingiverse.com/thing:4650640
1. Set up tree support with overhang angle of 70 degrees then slice. The support don't reach some model's spots.
2. Lowering the overhang angle value to 60 degrees makes the support reach those spots. 

**Screenshot(s)**
70 degrees
https://www.facebook.com/photo.php?fbid=10159523088174893&set=pcb.2806680066268827&type=3&ifg=1&__tn__=HH-R&eid=ARDMSYowORDDohrtl6fXEam3LLlt-c4GAGHwg-ZMALdR0z-32oE4x-bkEK-Wf3GP3xWi0_qaQKbQXeH0
60 degrees
https://www.facebook.com/photo.php?fbid=10159523088154893&set=pcb.2806680066268827&type=3&ifg=1&__tn__=HH-R&eid=ARAPkSC4dfAd-BvagtB2Nb3q5tkB1DI1FODlqHatLGsM9QzGvb0bcT1ADIgWm7X97zAYlZGU2Rmz2_Pd

**Expected results**
Should be the opposite results

**Project file**
.3mf file here
https://www.thingiverse.com/thing:4650640

Keep up the good work guys, Cura is the best slicer ever <3

konskarm, 2020-11-12:
Hi @VicOnTheBeach. Could you please post your pictures here? The Facebook links don't work.

VicOnTheBeach, 2020-11-12:
Hi Konstantinos,

Sure, here it is :)

Have a nice day

konskarm, 2020-11-12:
Since they don't show up properly when they are sent via email, I am uploading them for visibility:

70 degrees:
[image]

60 degrees:
[image]

konskarm, 2020-11-12:
The file is also not available in the given thingiverse link. Could you create a Cura project file (File -> Save project...) and upload it?

VicOnTheBeach, 2020-11-12:
[Bishop.zip](https://github.com/Ultimaker/Cura/files/5530692/Bishop.zip)

Ghostkeeper, 2020-11-17:
I'm not getting the same results with that project file. Here is the bishop at 70 degrees overhang (reduced support):
[image]

And here is the bishop at 60 degrees overhang (increased support):
[image]

It looks like the 60 degree overhang gets strictly more support than the 70 degree overhang.

Is this something that may have changed between the beta and stable? There were some changes to your printer's defaults in that time.

VicOnTheBeach, 2020-11-19:
Hi,

You're right, I've installed the stable release and the issue does not occurs anymore. Thank you for your investigations anyway.

Kind regards,

[image] [https://user-images.githubusercontent.com/2448634/99425731-85b7fb80-2903-11eb-9adb-166d6db516e4.png]
[image] [https://user-images.githubusercontent.com/2448634/99425997-e3e4de80-2903-11eb-956c-e2c5cedb6735.png]

closed 2020-11-19 by konskarm