Ultimaker/Cura #8740: object NOT over bed plate cura doesn't create support

bucefalo2, 2020-11-12:
**Application version**
4.8 

**Platform**
windows 10
**Printer**
Ender 3 pro

**Reproduction steps**
1. lifted item from bed plat of 2 mm in Z in cura 
2. enabled in SUPPORT -> GENERATE SUPPORT
3. SLICE  

**Screenshot(s)**
[http://www.simiweb.net/cura_problem.jpg](url)

**Actual results**
0 minutes printing time.. objected won't be sliced 
**Expected results**
by moving on Z by 2mm and generate support i'd expect cura to create a support to fill that 2 mm gap !

**Project file**
[www.simiweb.net/CE3_tyre_2.3mf](url)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**

nallath, 2020-11-12:
It's a known issue that adaptive layer height doesn't work when the object isn't touching the buildplate.

bucefalo2, 2020-11-12:
> It's a known issue that adaptive layer height doesn't work when the object isn't touching the buildplate.

i didn't know ! i made a test now DISABLED " use adaptive layers " and i could get the support , look : 
[www.simiweb.net/support.jpg](url)

Ghostkeeper, 2020-11-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/3183.
Slicing stops on the first empty layer there. That's a bug.

closed 2020-11-17 by Ghostkeeper