Ultimaker/Cura #8742: Cura 4.8 relative extrusion incomplete layers

dtwolfen, 2020-11-12:
**Application version**
Cura 4.8

**Platform**
Mac Catalina and Windows 10

**Printer**
Custom FFF

**Reproduction steps**
1. I slice model with absolute extrusion save gcode and print all good
2. I slice model with relative extrusion and certain layers are missing sections of the layers even though preview shows them. 
3. Print model and miss sections of layers.
4. Look at gcode in cura and see missing sections only have travel no extrusion. 

**Screenshot(s)**
[image]
[image]

**Actual results**
Incomplete layers when using relative extrusion

**Expected results**
Complete Layers when using relative extrusion

**Project file**

**Log file**

**Additional information**

[corner test.txt](https://github.com/Ultimaker/Cura/files/5531153/corner.test.txt)

dtwolfen, 2020-11-12:
Note this also happens with any other stl. Seems as though the top, bottom, and infill lines get traveled but no extrusion.

dtwolfen, 2020-11-13:
Find if I downgraded to 4.6 same thing, then downgraded to 4.4 and it worked. Something broke between 4.4 and 4.6. Sorry never used 4.5 so didn't have it available

rburema, 2020-11-17:
Hi @dtwolfen Since this probably depends on the (printer-)settings, could you still save (not export!) a project and attach it here? (As a zip, as gtihub is picky with file-formats, it basically only seems to accept zip, txt and images.)

This makes the debugging process easier as we don't have to second-guess ourselves.

Ghostkeeper, 2020-11-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/6174.

closed 2020-11-17 by Ghostkeeper