Ultimaker/Cura #8743: [4.8] Consistent and reproducible Crash Report

SilverThunderbird, 2020-11-12:
**Application version**
Cura 4.8

**Platform**
Windows 10 Home 64 bit, version 20H2

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Installed Cura 4.8, selecting uninstall previous version 4.7.1
2. Synced my settings with Ultimaker
3. Loaded a simple, previously successful file, sliced it
4. Tried to save it. Crash. The resulting saved file was corrupt. Happens every time, on any model.

Tried uninstalling/reinstalling several times, with same results.

**Screenshot(s)**
[image]

**Actual results**
Immediately after save, a Crash Report is generated, and the saved file is corrupted.

**Expected results**
No crash

**Project file**
Doesn't matter. Happened with 2 different small files that previously printed fine with 4.7.1

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5531288/cura.log)

**Additional information**
Plugins installed or displayed:
Cura Backups, FiPost Processing (show time remaining), Update Checker, Barbarian plugin, Mesh Tools, Auto Orientation, Settings Guide, Startup Optimizer, ThingBrowser, and Octoprint plugin.

Thank you.

rburema, 2020-11-17:
Hi @SilverThunderbird  I suspect that this is caused by https://github.com/Ultimaker/Cura/commit/6c9e414e8ad79e327dc1ac062d46f260339dbf8a which was part of an effort by a Tronxy user to update the Tronxy profiles.

As far as I can determine, it seems Tronxy now uses the 'normal' generic materials, instead of it's own. It seems that you still somehow have a reference to the old 'generic Tronxy pla' material, which in turn refers back to the original (un-updated) printer definition (which had another name as it was before that type (x5sa-400) was consolidated with several other definitions into a single 'Tronxy Pro' one).

You could try (re)setting the material to generic PLA (or if that doesn't work, maybe any other 1.75mm pla, they're all pretty close). That's just a workaround though.

At the very least it shouldn't actually _crash_ on this, I'll bring it up with my colleagues.

(The investigation started from this line in the logs: `UM.Settings.InstanceContainer.DefinitionNotFoundError: Could not find definition tronxy_x5sa_400 required for instance generic_pla_175_tronxy_x5sa_400
2020-11-11 22:25:42,599 - DEBUG - [MainThread] UM.Message._onInactivityTriggered [101]: Hiding message because of inactivity`.)

SilverThunderbird, 2020-11-17:
> that this is caused by 6c9e414

It was actually caused by a plugin that broke under 4.8.
As suggested, I removed "Startup Optimiser" which solved the problem.

rburema, 2020-11-17:
Alright, thanks for reporting back!

I'll hold off closing the issue until after I've discussed it though, as it's still a crash instead of it just not working.

SilverThunderbird, 2020-11-17:
Here's another thread with some more info:
https://github.com/Ultimaker/Cura/issues/8734#issuecomment-726864061

Thanks for your reply.

TheLastGriffin8, 2020-11-18:
anyone know how to fix this
[image]

Ghostkeeper, 2020-11-18:
If you're unable to remove the Startup Optimizer plug-in due to Cura not starting, you can also remove it from the `C:\Users\<username>\AppData\Roaming\cura\4.8\plugins` folder, just by deleting the folder of the plug-in there.

fvrmr, 2021-05-20:
Closing this issue, since it is removed of our backlog because our PO and QA-engineers believe we will not work on it anytime soon.

Ghostkeeper, 2021-06-04:
So it is not a priority then, but still a valid bug. We'll defer it.