Ultimaker/Cura #8745: [4.8] Selecting model not possible after switching to monitor tab

marine-3d, 2020-11-12:
Cura 4.8
Mac OS 11.0.1
2 x UM3's

With a model on the build plate, if you click on the monitor tab and then back to prepare/preview, model no longer selectable with mouse.

No difference if model just loaded and/or not sliced/sent to printer. Can be selected with keyboard shortcut/right click - select all models, but no interaction is possible.

nallath, 2020-11-13:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

marine-3d, 2020-11-13:
Hi Nallath,

Hopefully the attached video is enough to go on. Also, managed to isolate the issue to when the UM3 status is cloud or network connected (Blue icon with cloud or tick symbol). When the printers are off (grey icon), the issue isn't present and you can still select the model as normal, even when flicking between prepare/preview/monitor.

Hope this helps.

[Screen Recording 2020-11-13 at 16.48.21-1.mov.zip](https://github.com/Ultimaker/Cura/files/5538547/Screen.Recording.2020-11-13.at.16.48.21-1.mov.zip)

fieldOfView, 2020-11-13:
Does this also happen without the sidebargui plugin installed?

marine-3d, 2020-11-14:
Hi fieldOfView,

Just uninstalled the plugin but still the same issue.

fieldOfView, 2020-11-14:
~Wew, so it isnt my fault for a change.~ Err, I meant... shame that does not fix the issue.

scottmloving, 2020-11-16:
Same issue on my Macbook Pro with Ender 3. I just upgraded to Big Sur (11.0.1) and Cura 4.8.0. After monitoring a USB print, models are not selectable, even after starting a new project or reloading the models.When a model is selected from the object list, the move arrows appear, but they disappear as soon as they are clicked.

Ghostkeeper, 2020-11-17:
Hmm, I'm unable to reproduce the issue. I'm basically doing the exact same steps as marine-3d, just with a different model and my own UM3.

Do either of you have a log file you can share with us?

In my log I'm only seeing this warning when switching to the monitor page:

    plugins/UM3NetworkPrinting/resources/qml/MonitorStage.qml:75:13: Unable to assign [undefined] to bool

Seems that the `supportPrintJobsQueue` property is not defined for LAN printers. But that shouldn't really affect anything.

khaoskosmos, 2020-11-22:
I have this issue as well. MacOS Big Sur 11.0.1. If I go to the monitor tab and back, I can no longer select the model. Starting a new project does no clear the problem, only completely closing Cura (CMD + Q) and relaunching it.
[cura-no-select-macos.mp4.zip](https://github.com/Ultimaker/Cura/files/5578953/cura-no-select-macos.mp4.zip)

nikolafamit, 2020-11-23:
Same problem. Big Sur & Cura 4.8

marine-3d, 2020-11-24:
Log attached

[cura.log](https://github.com/Ultimaker/Cura/files/5590281/cura.log)

evtrados, 2020-11-25:
Hey all.
Just upgraded to Big Sur v11.0.1 from Big Sur beta, but I cannot reproduce the issue. I used .stl and .3mf files and 3 different models for each. I also tried removing the configuration folder, but still the issue is not there. Could it be model-specific? I doubt it, but in any case if you could send me the model with which you can reproduce the issue, or even better the whole configuration folder, I can try and check if it's appearing then. 
Thanks!

khaoskosmos, 2020-11-25:
It doesn't always do it, I'm not sure what triggers it. I am not signed in to Cura at all.

I just had it happen after leaving Cura open after a print. I imported a new stl and could not select it (you can get it to select by click on it from the object list in the bottom right, but you can't interact with it). Completely quitting Cura and reopening allowed me to select and manipulate the stl (same one as before). I could switch to the monitor tab, back to prepare with no issues. I was able to slice, reslice, change tabs, no problems. The printer was actually turned off during this though (it was also off in the first instance of Cura as well).

I turned my printer ON, switched to the monitor tab, back to prepare, and could not select the model again.
[cura.log](https://github.com/Ultimaker/Cura/files/5598138/cura.log)

SmithyAT, 2020-11-29:
I see the same issue here, Cura 4.8 and Big Sur 11.0.1.
First I was not able to reproduce it, a user in the forum had the problem, so I looked deeper on it on my system. It has nothing do to with the monitor tab, which I don't use very often. It occurs when I leave Cura open for some time. For example I sliced and printed something on Friday and didn't close Cura. Today (Sunday) I tried to select the part on the build plate I printed on Friday and I was not able to do it anymore. Between Friday and Sunday, I did nothing with Cura.

I think you should be able to reproduce it when you just do nothing for one or two days.

nikolafamit, 2020-11-29:
> I see the same issue here, Cura 4.8 and Big Sur 11.0.1.
> First I was not able to reproduce it, a user in the forum had the problem, so I looked deeper on it on my system. It has nothing do to with [...]

Yes. The same situation happened to me. Although, it wasn’t for days but in my case Cura was opened for 6-7h

fieldOfView, 2020-11-29:
> just do nothing for one or two days.

Yeah, that's not going to happen in a hurry ;-)

Gatix, 2020-11-29:
So I have the same issue, Cura 4.8 and Big Sur 11.0.1. I'm on a Macbook Pro with an external monitor.

If the Cura window is on the external monitor, I can't select object nor click on the axes. Moving the window to the laptop screen fixes the issue. Perhaps the difference in pixel density or resolution is causing the issue?

SmithyAT, 2020-11-29:
Interesting observation that moving the window to the laptop fixes the issue.
I can reproduce it on my iMac with default resolution but have not seen it yet on my MacBook Pro.

evtrados, 2020-11-30:
Developers, see internal issue CURA-7897.

meichthys, 2020-11-30:
I can also confirm that the issue happens on my Macbook Pro (Big Sur) without any external peripherals attached.

meichthys, 2020-11-30:
I can also reproduce the issue every time using the following steps:

1. Open Cura
2. Add stl to build plate and select model
3. Click the following menu icon:
[image]
4. Click/Drag on the build plate (Not on a model)
At this point the models bill become un-selectable.

SmithyAT, 2020-11-30:
Bingo, with these steps I can reproduce it easily.

nallath, 2020-12-01:
We've also been able to reproduce it (at least) once on our side. I still don't quite get how this is only happening with big Sur though.

SmithyAT, 2020-12-01:
Thanks Jaime for the update. If there is anything I can do or help, just let me know.

Rukongai, 2020-12-02:
I get this one a lot. I have to restart the application every 15-20 minutes. Is there any logs I can provide to help track this down?

nallath, 2020-12-02:
I'm not the product owner, so I don't decide the priority of issues. But I (and our QA engineers as well) have pushed for this to get priority. As it stands I expect that this will be picked up for the 4.9 release. 

 Right now it looks like that we have good/reliable reproduction steps, which is usually enough to fix the issue.

GuyJanssen, 2020-12-02:
I have exactly the same problem and it popped up after Big Sur was installed and I still had Cura 4.7. It is not solved by moving the window from my external screen back to my MacBook screen.
The problem rises when I add per model settings or even just click on the icon. I use per model settings a lot so I practically have to restart Cura for every object which is pretty often in a 3D printing business. Hopefully you will solve this in 4.8.1 and not in 4.9!

yzorgone, 2020-12-03:
i have the problem too since Big Sur.

timonwehmann, 2020-12-07:
I have the same problem as well. Occurred also on Catalina, but not as predictable as it manifests on Big Sur.

fabiopdm, 2020-12-08:
Using some heavy 3D Models, I have the same issue since Big Sur. Using the last version of Cura, 4.8.0.

When I click to slice or sometimes when I ask to rotate de model to put the face on the plate the Cura software freezes.

gzumwalt, 2020-12-08:
On my iMac running Big Sur, I can only click on a .STL one time to rotate / translate / etc.  After the first time, I can no longer select the .STL.

Also, the preview mode is a quite screen.

xking3, 2020-12-11:
Same problem here since Update to Big Sur.

nuaimat, 2020-12-11:
As a walkaround, I downgraded to 4.7.1, hope this bug will be resolved soon.

GuyJanssen, 2020-12-11:
On Big Sur version 4.7.1 has the same problems

nuaimat, 2020-12-11:
@GuyJanssen 4.7.1 worked for me on Big Sur, yes sometimes I have to select objects by clicking on their name in the "Object list" on left-bottom, but I was able to add all needed no-support areas of https://www.thingiverse.com/thing:4239135 and slice it. I couldn't do the same on 4.8

GuyJanssen, 2020-12-12:
thanks nuaimat, it still is a pain in the ..... really hope they will fix this very soon ........

bkarr123, 2020-12-26:
I can confirm I am having the same issue. I just updated to Bug Sur 11.1, in Cura 4.8.0 I am having trouble moving  around models, scale, rotate, or anything else. Please help!

Salty-Doug, 2021-01-02:
Same Issue here with a MacBook Pro mid-2014 and Big Sur Cura 4.8.0.
Here is my log file:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/5760808/cura.log.zip)

rhukster, 2021-01-04:
Jumping on the dog pile... restarting cura is quickfix but getting annoying!

GuyJanssen, 2021-01-05:
Hi Ultimaker, is there any sight on a solution??? It is getting pretty annoying running a couple of Ultimaker printers like this!!

Thalioss, 2021-01-05:
I have the same issue. REALLY annoying to have to quit and restart Cura every time I send a job to a printer.

fieldOfView, 2021-01-07:
Could you try turning on the `Force layer view compatibility mode (restart required)` option in the General preferences?

GuyJanssen, 2021-01-07:
> Could you try turning on the `Force layer view compatibility mode (restart required)` option in the General preferences?

I did try that but that doesn't help!

fieldOfView, 2021-01-07:
> I did try that but that doesn't help!

And - just to be sure - you did restart Cura so the setting had effect?

GuyJanssen, 2021-01-07:
Yep. After turning it on I restarted and loaded a multi-object 3mf project and could not even select the objects. After turning the flag back off , restarting and loading the same 3mf I could select the separate objects.After clicking the settings per object icon on the left I could not select the objects anymore. So still the same problem!

Lukapp, 2021-01-08:
Same problem. I can reproduce it without having to restart the application. It is blocked only with the support icon that you comment.

But I have a plugin installed, "Custom Printjob Naming", which allows me to unlock the model without having to restart. This plugin allows us to save some characteristics we are interested in the filename, but it is not necessary to use it, it is only to open a pop-up window. Maybe works with another plugin that opens a popup window.

Unblock
1. "Custom Printjob Naming" plugin installed or "maybe" another plugin that opens a popup window.

2. Select a model with Command-A and select any other button (scale, move), not the problematic icon.

3. Open the plugin properties window by clicking on the name to save. Cancel or OK.

4. Select the model again and it is already unlocked.

https://user-images.githubusercontent.com/8763220/104058236-b590ca80-51f3-11eb-95e5-bcf7a71c4212.mov

 I don't know if it will work with other plugins that open a popup.

Maybe the icon function loses focus or does not return it, and freezes, but with this external plugin or others when you open a window and close, it regain focus of the window, I don't know. But the problem must be in the function of the icon.

[image]

GuyJanssen, 2021-01-09:
hi @Lukapp, works like a charm!! Thanks so much for your workaround!!  Will save me a whole lot of time!!
@fieldOfView Now Ultimaker hopefully knows what to do about it!

Ghostkeeper, 2021-01-12:
It's not a solution for us to open up a pop-up every 10 seconds or so to unlock scene interactions in case they have gotten stuck recently.

What we know so far is that apparently some elements are still catching mouse events even though they are invisible, but only on MacOS Big Sur.
We also know that Big Sur is not yet supported in Qt5 either: https://doc.qt.io/qt-5/supported-platforms.html This is probably why it's giving so much problems for us. Qt's own developers are not confident enough with Big Sur bugs that they'd call it "supported", even though they've been working on fixes here and there. Aside from that, Cura is still on a fairly old version of Qt as well: 5.10. This has been necessary so far to be able to support MacOS 10.11 (El Capitan). Since we see El Capitan's usage dwindling we're considering updating Qt to 5.15. That's about everything we can really do about this issue, so that's the plan.

uid01089, 2021-01-13:
I faced the same issue on Mac OS Big Sur. As workaround, I
1. Open a stl-file, the object is then on the plate
2. Save it as a Cura Project
3. Close Cura and re-open it
4. Load the saved Cura Project

Then I was able to modify the object.

Einstein2150, 2021-01-26:
This issue is really old. I had this issues before Big Sur too - maybe not so often but after a standby of my macbook every time ...

mbourque, 2021-02-28:
+1 I have this problem too. Awaiting a fix.

lewsid, 2021-03-06:
Throwing another hopeful whimper onto the pile, this is a workflow-breaking bug. Happens 100% of the time after saving anything. Constantly closing and re-opening the app is a bummer of a work-around, those delays add up.

Ghostkeeper, 2021-03-08:
We've just made a testing build with Qt 5.15 and confirmed that this indeed fixes the Big Sur issue. Progress with the upgrade of the Qt version is still ongoing.

enegron, 2021-03-08:
uid01089 comment describing "save as project" workaround is working for me, however I assume this will be fixed since the rotate feature is available and can be activated after loading an STL file.

colomonkey, 2021-03-11:
I have the same problem after saving to sd card. The models cannot be selected anymore and the only way to proceed is to restart Cura. Using  Cura 4.8.0 on latest version of Macos Catalina.

dJOS1475, 2021-03-16:
I have this issue on 4.8 running on MacOS 11.2 also - I'm getting really sick and tired of having to quit Cura and re-run it every time this happens! 👎 

This bug has been present for months, please fix it already!

see3d, 2021-03-19:
I also have this problem with my Mac Mini running Big Sur.

fasihrana, 2021-04-01:
Same issue. The preview window is also blank if I slice after opening any dialogue boxes.

jamiebikies, 2021-04-01:
Same issue for me 🖐️

holstvoogd, 2021-04-03:
I can reproduce this consistently in 1 easy steps: go to monitor tab (with or without a model loaded even).

Now both the prepare and preview modes are broken.
I use the sidebare GUI pugin & Octroprint.

The logs do contain a bunch of errors related to both:
```
2021-04-03 15:37:48,886 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
2021-04-03 15:37:48,918 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/MonitorStageMenu.qml:31: TypeError: Cannot read property 'children' of undefined
2021-04-03 15:37:49,510 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-03 15:37:52,775 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PrepareStage
2021-04-03 15:37:52,801 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2021-04-03 15:37:52,909 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
2021-04-03 15:37:52,913 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
2021-04-03 15:37:52,917 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
2021-04-03 15:37:52,921 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/SidebarGUIPlugin/SidebarGUIPlugin/resources/qml/ViewOptionsPanel.qml:148:13: QML QQuickItem: Binding loop detected for property "children"
2021-04-03 15:37:52,934 - DEBUG - [MainThread] cura.CuraApplication.log [1721]: SidebarGUIPlugin patching interface for Cura 4.7 and newer
2021-04-03 15:37:52,944 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/Menus/ConfigurationMenu/CustomConfiguration.qml:234: TypeError: Cannot read property 'activeStack' of undefined
2021-04-03 15:37:53,006 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:88: ReferenceError: stop is not defined
2021-04-03 15:37:53,011 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:72: TypeError: Cannot read property of null
2021-04-03 15:37:53,015 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:156: TypeError: Cannot read property of null
2021-04-03 15:37:53,022 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:167: TypeError: Cannot read property of null
2021-04-03 15:37:53,028 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:168: TypeError: Cannot read property of null
2021-04-03 15:37:53,034 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:204: TypeError: Cannot read property of null
2021-04-03 15:37:53,039 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:205: TypeError: Cannot read property of null
2021-04-03 15:37:53,045 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:118: TypeError: Cannot read property of null
2021-04-03 15:37:53,050 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:185: TypeError: Cannot read property of null
2021-04-03 15:37:53,054 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Users/arthur/Library/Application Support/cura/4.8/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/MonitorItem4x.qml:187: TypeError: Cannot read property of null
2021-04-03 15:37:53,057 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
[... cut ...]
2021-04-03 15:37:53,258 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/MonitorButton.qml:231: TypeError: Cannot read property of null
2021-04-03 15:37:53,262 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/MonitorButton.qml:241: TypeError: Cannot read property of null
2021-04-03 15:38:01,220 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///Users/arthur/Projects/3D Printers/Voron-2/STLs/TEST_PRINTS/Voron_Design_Cube_v7.stl
```

Ghostkeeper, 2021-04-08:
The day before yesterday we released a beta of Cura 4.9: https://github.com/Ultimaker/Cura/releases/tag/4.9-beta

In our own findings, this issue is fixed with this beta. Would you like to test as well?
There is still an issue with some drop-downs in that beta, which is specific to MacOS. But it's usable at least. We're trying to fix those up for the stable release.

[12 more replies]

closed 2021-04-29 by Ghostkeeper