Ultimaker/Cura #8746: Auto centering and drop on plate

ShiroKrow, 2020-11-12:
**Application version**
4.8.0
**Platform**
Windows 10/ Nvidia
**Printer**
Anet A6
**Reproduction steps**
1. Ctrl + R
2. Or just arranging models

**Screenshot(s)**
No need to picture something that doesn't work

**Actual results**
Arranging models doen't work any more, it centers the piece but does not drop it on the plate

**Expected results**
Before this version the piece was automatically dropped on the plate

**Additional information**
It happens since the 4.8.0

konskarm, 2020-11-13:
In your preferences, "Automatically drop models to the buildplate" should be checked. If it is, Ctrl+R will arrange the models and drop them on the buildplate. If its not, Ctrl+R will arrange the models but keep them in their current Z position. In any case this is working as expected (as far as I have tested so far with your reproduction steps), so I am closing this.

ShiroKrow, 2020-11-13:
I'm sorry but in the 4.7 the Ctrl + R could reset the Z position. If I check this option I can't manually set Z position the way I want anymore, which means it's either I check the option or I keep the means  to set Z position to my liking like I did before without problems. If you can't do anything about that then I'll just go back to 4.7 then.

konskarm, 2020-11-13:
Since 4.8, we are using a new arrange algorithm so the behavior has changed considerably. I'll discuss with the team about making arrange also drop models on the buildplate.

ShiroKrow, 2020-11-13:
Alright then thank you !

Ghostkeeper, 2020-11-17:
We've discussed this yesterday, but decided that the arrange function is not expected to drop models to the build plate if the "drop to build plate" preference is enabled. That would be fairly annoying for meshes that are vertically aligned.
If this happened in 4.7, then that was a bug. Sorry! If you want to have it drop to the build plate, you'd need to either set the Z coordinate to 0 manually or enable the preference. There are other threads open to make this user experience better (with perhaps an entry in the context menu to drop the selection to build plate or something).

closed 2020-11-17 by Ghostkeeper

ShiroKrow, 2020-11-17:
It was already there since the 4.3 when I started and maybe before so your "bug" was quite useful to many since the "drop to build plate" option doesn't seem useful now that I can't drop on build plate with Ctrl+R after setting orientation to a face, manually setting the Z coordinate isn't a solution since the coordinates' origin will never jump to the lowest position of a part when changing orientation. I need the command to do as it did so I'll stay with the 4.7.0, but thank you anyway.

mak0t0san, 2020-11-17:
I'm having issues in 4.8 where using the "Lay flat" or "Select face to align to the build plate" options are not working correctly.    I don't have the "Automatically drop models..." preference enabled, but would expect using the options to place it on the bed would work correctly.

Ghostkeeper, 2020-11-19:
> manually setting the Z coordinate isn't a solution since the coordinates' origin will never jump to the lowest position of a part when changing orientation.

It's expected to. The Z coordinate in the move tool is always the position of the lower end of the Z range of the bounding box of the model. If setting the Z coordinate to 0 doesn't work to align the bottom of the mesh with the build plate, that would be a bug!

Ghostkeeper, 2020-11-19:
> I'm having issues in 4.8 where using the "Lay flat" or "Select face to align to the build plate" options are not working correctly. I don't have the "Automatically drop models..." preference enabled, but would expect using the options to place it on the bed would work correctly.

Indeed, I can reproduce that, and with Lay Flat at least I'd say that the expected behaviour is that it should drop the model down. I'll bring this up as well.