Ultimaker/Cura #8747: no automatic sign-in and always what's new screen

DreadKnight, 2020-11-12:
4.8.0 is problematic for me, as it was giving me the starting set-up screen each time, even if I completed it before, then was forgetting to open the file. I'm on linux, removed ./config/cura and .config/Unknown organization/cura.conf
After that I've went through the set-up again, logged in, removed the duplicated Creality CR-6 SE that was added each time, synched account and restarted app. It never signs in automatically and I keep getting the "what's new" screen.

nallath, 2020-11-13:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

DreadKnight, 2020-11-13:
@nallath Yes, my bad. I tend to delete template first thing, realized after I should have used it around. Will get to implementing it.

DreadKnight, 2020-11-13:
@nallath Alright, template implemented. Looking for some feedback.

nallath, 2020-11-13:
I'd still like to get the log files. 

I guess that it can't save the preferences file for some reason (which is the file that stores the login token & what the last run version was). 

What I can't quite explain is why removing the ./config/cura/ folder doesn't help, since that is the place where the preferences are stored.

DreadKnight, 2020-11-13:
@nallath Was thinking the exact same thing, hence why I removed those.
I wanted to provide logs, but only found info on how to fetch that when on Windows.
Will have another look at it soon as I was in quite the rush and might have missed obvious info perhaps.

DreadKnight, 2020-11-13:
Well, I've started it from console, this is what I got just for app launch:

>dread@FreezingMoon:~/Applications$ ./Ultimaker_Cura-4.8.0_6782abb3abec599bf7274297cdaa94bb.AppImage 
Fontconfig warning: line 5: unknown element "its:rules"
Fontconfig warning: line 6: unknown element "its:translateRule"
Fontconfig warning: line 9: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 8: unknown element "description"
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 76: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 76: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 84: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-Hack.conf", line 17: invalid edit mode "delete"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-fonts-persian.conf", line 34: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/65-fonts-persian.conf", line 35: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 4: unknown element "its:rules"
[... cut ...]
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 5: unknown element "its:translateRule"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range

fieldOfView, 2020-11-13:
> I wanted to provide logs, but only found info on how to fetch that when on Windows.

See https://support.ultimaker.com/hc/en-us/articles/360016538000-Saving-Ultimaker-Cura-logs

> I've started it from console, this is what I got just for app launch:

Those are not the logs we are looking for.

DreadKnight, 2020-11-13:
Alright, managed to miss that folder apparently via KDE's kinda broken searches, would have deleted it as well.

I've pasted the thing over here, because it's quite big: https://gist.github.com/DreadKnight/0fc3bd14ec032b966f1d00958b96e360

nallath, 2020-11-13:
uh. Something is causing an upgrade though...
```

2020-11-13 16:27:45,510 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2020-11-13 16:27:45,526 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [336]: Create upgrade task for configuration file [./cura.cfg] with type [preferences] and source version [6000015]
2020-11-13 16:27:45,534 - INFO - [MainThread] UM.VersionUpgradeManager._upgradeFile [421]: Upgraded ./cura.cfg to version 6000016.
2020-11-13 16:27:45,536 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [175]: Checking and performing updates took 0.02736687660217285

SNIP

2020-11-13 16:27:48,590 - WARNING - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [119]: Could not retrieve token data from auth server: {"errors": [{"id": "14f9be11-ce7b-49f3-aba8-0584a229beb5", "code": "invalidToken", "http_status": "403", "title": "The given token is invalid.", "meta": {"error_type": "AuthError", "url": "https://account.ultimaker.com/check-token"}}]}
2020-11-13 16:27:48,594 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token.
2020-11-13 16:27:48,813 - WARNING - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [112]: Unable to use the refresh token to get a new access token.
2020-11-13 16:27:48,816 - WARNING - [MainThread] cura.OAuth2.AuthorizationService.getUserProfile [86]: The user profile could not be loaded. The user must log in again!
2020-11-13 16:27:48,819 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data
```

DreadKnight, 2020-11-13:
I see. Well, deleting the `~/.local/share/cura` folder and redoing the initial set-up fixed the issue for me, but not a smooth experience compared to the other versions and KDE's fail search sure didn't help (been very problematic for a while now), oh well...

HarlemSquirrel, 2024-02-28:
This worked for me as well. I moved the folder to the trash, set up Cura again, then closed it, and copied all the sub folders back over excluding the `cura.cfg` file because when I copied that over and started Cura I got the setup screen and I was logged out so I think this file is the issue. It looks like Cura 5.6 doesn't create this file anymore so maybe it was confused when trying to "upgrade".