Ultimaker/Cura #8748: Start / End gcode displayed as boxes instead of chraracters starting between 3.5.1 and 4.8.0

mtfurlan, 2020-11-12:
**Application version**: 4.8.0

**Platform**
* Debian 10
* kernel 4.19.0-12-amd64
* 8 core Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz
* GeForce RTX 2070

**Printer**
* ultimaker 2

**Reproduction steps**
1. Download cura
2. rm -rf ~/.local/share/cura/*
2. Open cura, setup printer
3. look at machine settings

**Screenshot(s)**
[image]

**Actual results**
It's a display issue, I can copy the gcode out and it works just fine.

**Expected results**
I expect to see the actual gcode.

Cura 3.5.1 does not do this.

**Log file**
http://ix.io/2DWk
The log for 3.5.1 and 4.8.0 are identical.

If this can't be reproduced and someone is interested in fixing it I can try a few versions in between.

rburema, 2020-11-17:
Hi @mtfurlan This is a duplicate of #7836 . Per policy, we close duplicates in favour of the original issue (which is still open).

It seems to be caused by machines having newer font libraries compared to our (old!) build system. We've been looking to upgrade the build-system for a long while now, but given the requirements, progress has been slow.

See that ticket for (further) discussion.

closed 2020-11-17 by rburema

mtfurlan, 2020-11-17:
Ah, thank you!
I had searched for "box" not "square".

rburema, 2020-11-17:
No problem, it happens a lot!