Ultimaker/Cura #8750: [4.8] Superfluous unsupported infill

Epicurusismyhomeboy, 2020-11-12:
**Application version**
4.8

**Platform**
Win10

**Printer**
Ender 3

**Reproduction steps**
When slicing a specific model in 4.8, 21 layers in I get two layers of unsupported infill across the entire model despite there being no geometry that requires it. If I increase my wall count from 2 to 3, those two layers of unsupported infill disappear. This cannot be reproduced in 4.7.1 or 4.6.

**Screenshot(s)**
Cura 4.8
[image]
Cura 4.7.1
[image]

**Actual results**
Terrible terrible things happen.

**Expected results**
There should be no unsupported infill.

**Project file**
[CE3_Ulvheim_Building_roof.zip](https://github.com/Ultimaker/Cura/files/5533614/CE3_Ulvheim_Building_roof.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5533621/cura.log)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-11-13:
Indeed, it reproduces with your model, but I am closing this as a duplicate of https://github.com/Ultimaker/Cura/issues/6728.

closed 2020-11-13 by konskarm

konskarm, 2020-11-13:
This is now in our backlog. Devs see internal ticket CURA-7845.