Ultimaker/Cura #8751: Timer for TFT screen in pause-at-height script

robblot, 2020-11-13:
The Pause at height post processing script, the BQ(M25) setting, works for my Sidewinder X1 but only for a few seconds because of the TFT screen. I want to use this feature to add a magnet to my print and to also do a filament change during the print. I am running the advanced Marlin version 1.1.9 by Earl Miller along with his TFT firmware.

If a timer could be implemented so I could have enough time to do the filament change and then resume the print. A setting could be added to the script and allow the user to set a duration for the script to run before the print resumes.

Right now it seems my only option is to use Simplify3D and create separate profiles.

This would greatly benefit anyone using a printer with a TFT screen instead of an LCD screen

Ghostkeeper, 2020-11-17:
I don't see how the type of display would interact with the pause functionality at all. Should it not display roughly the same content on the display, regardless of which technology the display uses to emit patterns of light?

What g-code would Cura need to write in order to implement a timer? `M25` has no parameters afaik.

robblot, 2020-11-17:
In the current script there is an M25 option which is in Marlin. I think it activates an M125 implementation in Marlin taking the options supplied in the cura script. All of that works, but I had no control of when of the pause would resume. After a few seconds the print would resume leaving no time to do anything to the print.

Ghostkeeper, 2020-11-19:
`M25` by itself is not documented to have any parameters: https://marlinfw.org/docs/gcode/M025.html

`M125` does take parameters, but not a pause duration, since it just moves the head and doesn't pause it: https://marlinfw.org/docs/gcode/M125.html

Looking beyond the documentation, indeed you're right that `M25` calls `M125`: https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/gcode/sd/M24_M25.cpp#L86

You can also see that it takes an undocumented `P` parameter: https://github.com/MarlinFirmware/Marlin/blob/072b15784f79c51cf4aed9b6272dfdba8160b452/Marlin/src/gcode/feature/pause/M125.cpp#L78 which is interpreted as a boolean. But no pause duration.

I don't see a way in which we can really help, here.

robblot, 2020-11-19:
Ok, thank you for your time.

closed 2020-11-19 by robblot

ryanaxiom, 2021-02-12:
I know this was marked as closed but I'm posting this just in case someone else finds this via web search or robblot is still looking for a solution for the Sidewinder X1.

Robblot is likely correct that the TFT is causing the incorrect pause functionality.  With the stock firmwares (TFT and mainboard), Artillery added a manual pause button which prevents normal functioning of any of the pause commands in Cura (pause at height, color change, etc).  The pause command works exactly as robblot describes where the printer parks in the pause position for about 10-20 seconds and then simply proceeds on with the print.  It may have to do with the way the TFT pause button was implemented since the USB and SD slots are actually included on its board.  There are a couple of workarounds that I've found.

1) Update TFT and mainboard firmware.  See https://www.thingiverse.com/thing:4294049 .  I have not done this yet, but plan to do so very soon.  Many have had success and digant is maintaining this.  The instructions are in the Thing Details and you need to update the TFT according to those instructions before you work on the mainboard.  There is a way to update the mainboard without opening the printer (you can search that online) but it seems to have varying success in terms of getting it to start updating the mainboard firmware.  This is due to the communication pathway of the TFT and mainboard.   Once the TFT boots, the firmware update on the mainboard is disabled.

2) Plane cut and print separate files one after another.  Make sure on the other files to include a part (small throwaway cylinder perhaps) that will raise the z level to at least what is required.  You may also need to disable the "delete empty layers" option (but if you have the cylinder on the plate that shouldn't be necessary).  This is needed because it will start from home and go directly to the start of the higher layer possibly hitting and dislodging the previously printed object.  Other disadvantage: you have to do this soon after the first print finished because it does not maintain temperatures (essentially this is like a cold pause).

3) Use the filament runout sensor to cause a hot pause.  This is better and can be accomplished as in this video.  https://www.youtube.com/watch?v=y_WvqhpclNc&ab_channel=Rich%27s3DArtilleryprinting .  The advantage of this is that the pause will be a hot pause so the plate will stay warmed.  The disadvantage is that you cannot use the filament runout sensor as intended when doing this.

So 1 is the likely the best option but requires firmware updates and likely opening the printer to do so.  2 works without modification, but you have to manually intervene soon to keep the bed hot and make sure the second print is set not to crash into the first (by adding a waste object).  3 will give warning beeps but requires some setup and you lose the filament runout sensor for that particular print.

Ghostkeeper, 2021-02-19:
In #9272 we've also put a workaround, basically still adding a pause at height, but then using a search and replace to replace all `M0` strings with a delay command `G4 S20`. That would cause it to wait for 20 seconds instead of the dysfunctional pause command.