Ultimaker/Cura #8752: Cura 4.8, Object not apearing only shadow

PotatoLord3, 2020-11-13:
**Application version**
4.8

**Platform**
win10

**Printer**
Ender 3

**Reproduction steps**
After updating to 4.8 from 4.7, and opening a .stl made in Blender. which worked on 4.7, but in 4.8 only shadow appeared with no object, also it doesn't allow to slice the object. It appears that the object appears somewhere beyond the build area. 

[image]

[image]

nallath, 2020-11-13:
Please attach the STL that causes this issue.

PotatoLord3, 2020-11-13:
> Please attach the STL that causes this issue.

I attached the .stl now. The problem is mostly that the STL object is spawned somewhere in the void beyond the build area.

nallath, 2020-11-13:
When trying to load this from master I'm getting an invalid file warning. I know that we recently added some checks for corrupted models that could cause crashes in some cases.

PotatoLord3, 2020-11-13:
> When trying to load this from master I'm getting an invalid file warning. I know that we recently added some checks for corrupted models that could cause crashes in some cases.

Ok, reuploaded it so it shouldn't be corrupt now. It worked on my pc fine so the file is not corrupted as far as I know.

GregValiant, 2020-11-13:
Message on load:  "Auto scaled to 100000% of original size"
Message on load:  "Model is not watertight"
I could see the shadow on the build plate.
Right clicked on build plate and hit "select all models"
Select Move tool.  It showed the model at X=0, Y=0, Z=29883.6119
Changed Z location to 0
"Check Meshes" returns - Not watertight and consists of 24 sub-meshes
"Analyze Meshes"  689777 vertices and 1379456 faces
"fix simple Holes"  Cura worked for 7 minutes
The model did slice.

[image]

Ghostkeeper, 2020-11-13:
Note that Blender by default works in metres, and subsequently also saves the model in metres. If you're modelling something in real sizes in Blender, be sure to first set your scene's units to millimetres.

thedaviator, 2020-11-15:
[CE3_WHITE_-_HULL_bigger_holes.zip](https://github.com/Ultimaker/Cura/files/5541754/CE3_WHITE_-_HULL_bigger_holes.zip)

Same issue here. Never happened in older versions of Cura. STL exported from Blender 2.81 in actual size in millimeters (not meters). 
Occasionally, Cura 4.7 (and earlier) thinks my STL is 10X larger than it is while other times it gets it right. 

Now with Cura 4.8, ALL of my Blender STLs show up as a shadow and cannot be sliced. `

Ghostkeeper, 2020-11-17:
We've added a ticket for this on the backlog. Developers, please refer to CURA-7855.

> Occasionally, Cura 4.7 (and earlier) thinks my STL is 10X larger than it is while other times it gets it right.

This is the preference "scale extremely small models" at work. Cura tries to detect the unit of the STL file if it is extremely small, precisely because of the aforementioned issue where some CAD software exports the model coordinates in metres instead of millimetres. What constitutes "extremely small" depends on the diagonal length of the build volume.

korealex, 2021-01-15:
any workaround?

nallath, 2021-01-15:
This is fixed by https://github.com/Ultimaker/Cura/pull/9128