Ultimaker/Cura #8757: Mistake in generating wipe code at start

YaronSoffer, 2020-11-13:
The following code is generated by cure :
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 **Y200.**0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 **Y200.0** Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line

My bed center ( X0,Y0)  in the middle of the bed  ( not at some corner as in  REPRAP Duet3d).
This causes a problem of going over the edge of the bed ( and actualy swiping the PEI sheet of it).
This should be 
G1 X0.1 **Y100.**0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 **Y100.0** Z0.3 F5000.0 ; Move to side a little

This can be fix in the "Post processing" -> "Manage Gcode"  -> "find and replace". 
Replacing the bad line with the correct one.
However, this should be fixed in the autogenerated code.

nallath, 2020-11-13:
This isn't code that is (auto)generated by Cura, this is code in your start-end g-code. 

You can change this via the machine settings (preferences -> printers -> machine settings

closed 2020-11-13 by nallath

YaronSoffer, 2020-11-13:
Hi Thanks - you are right . 
Did the modification there.