Ultimaker/Cura #8759: Zichzack-Elemente Stützstruktur verbinden läßt sich nicht deaktivieren

Krasi50, 2020-11-13:
Die Stützstruktur wird immer verbunden obwohl es deaktiviert ist. 
Cura 4.8.0
Windows 8.1 pro 
 64 bit
Ultimaker 3

[image]

nallath, 2020-11-13:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

Please provide a bug report in English and provide a project file that exhibits the issue.

Krasi50, 2020-11-13:
forget it - i will use simplify ...

nallath, 2020-11-13:
Okay. I'll close the issue then.

closed 2020-11-13 by nallath