Ultimaker/Cura #8761: skin fill option

viper0078, 2020-11-13:
Simplify3D has a option called "Allow single extrusion fill".
This option allows you to print with different amounts of filaments in a small area, as shown in the image on the right.
[image]
This has the following effect.
1) Shorter printing time.
2) Reduction of the load on the mechanical system.

Please consider this.

Ghostkeeper, 2020-11-17:
The terminology is confusing here. With "single extrusion" they mean a straight line instead of the zig-zag pattern if the shape is thin enough.

I'll see what my colleagues think of this. We have something similar [in progress](https://github.com/Ultimaker/CuraEngine/pull/1210) which has a bit more flexibility too, but it won't transform the infill patterns like this. It could be surprisingly complex because the width of a polygon is somewhat ambiguous in the general case.

bishop-5, 2020-11-19:
In the image on the right, we are deforming the Solid Layer instead of Infill.
This corresponds to the Skin of Cura, I think.

As stated in the parameters, a line can be filled with a single line if it corresponds to between 30% and 200% of the line width (the user-set value).

I see that a similar feature is under development in Cura.
I'm looking forward to it.

viper0078, 2020-11-19:
I commented on a different computer, so it's now someone else's ID. I'll rewrite it.
------------------------------------------------
In the image on the right, we are deforming the Solid Layer instead of Infill.
This corresponds to the Skin of Cura, I think.

As stated in the parameters, a line can be filled with a single line if it corresponds to between 30% and 200% of the line width (the user-set value).

I see that a similar feature is under development in Cura.
I'm looking forward to it.

Ghostkeeper, 2020-11-30:
(Developers, refer to issue CURA-7893.)

Ghostkeeper, 2021-07-07:
CURA-7893 was closed because the issue is fixed by #1210. You could try it out with the Arachne beta release we released in March.

Ghostkeeper, 2021-12-14:
This is implemented in https://github.com/Ultimaker/CuraEngine/pull/1210. The width at which it starts to coalesce into one line is defined by the line width and the join threshold.

closed 2021-12-14 by Ghostkeeper