Ultimaker/Cura #8762: ONE AT A TIME mode will not slice in 4.8

ScottKarch, 2020-11-13:
**Application version**
4.8.0

**Platform**
OSX

**Printer**
Ender-3 Pro

**Reproduction steps**
1. layout 12 PPE mask straps very close together
2. configure with print "one at a time"
3. Attempt to slice and it fails

**Screenshot(s)**
[image]

**Actual results**
attempting to slice
[image]

**Expected results**
Opening same saved file with 4.7.1
[image]

and slicing works
[image]

**Project file**
[8 ABS strap each x12.gcode.zip](https://github.com/Ultimaker/Cura/files/5538070/8.ABS.strap.each.x12.gcode.zip)

**Log file**
KEY LINES
2020-11-13 10:35:17,961 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [246]: Starting to slice...
2020-11-13 10:35:17,961 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [264]: Going to slice build plate [0]!
2020-11-13 10:35:18,798 - WARNING - [Thread-2] CuraEngineBackend.StartSliceJob.run [224]: No objects suitable for one at a time found, or no correct order found

Load file and slice
2020-11-13 10:35:07,653 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///Users/administrator/Desktop/3D Models/Covid Work/ Straps/8 ABS strap x12.3mf
2020-11-13 10:35:08,097 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.002920866012573242 seconds
2020-11-13 10:35:08,130 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.022085189819335938 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,161 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.0019450187683105469 seconds
2020-11-13 10:35:08,201 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.027032852172851562 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,224 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.0018951892852783203 seconds
2020-11-13 10:35:08,262 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0263369083404541 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,343 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.006643056869506836 seconds
2020-11-13 10:35:08,388 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03583097457885742 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,414 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.003662109375 seconds
2020-11-13 10:35:08,458 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03527998924255371 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,482 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.008157968521118164 seconds
2020-11-13 10:35:08,528 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.036457061767578125 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,560 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.01152491569519043 seconds
2020-11-13 10:35:08,604 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03407096862792969 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,637 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.007951021194458008 seconds
2020-11-13 10:35:08,690 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.04292702674865723 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,739 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.012251138687133789 seconds
2020-11-13 10:35:08,788 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03500699996948242 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,809 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.003454923629760742 seconds
2020-11-13 10:35:08,848 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.029835939407348633 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,874 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.005443096160888672 seconds
2020-11-13 10:35:08,905 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking /dev/cu.usbmodem1A13501 if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
2020-11-13 10:35:08,914 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03324699401855469 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:08,942 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.008502006530761719 seconds
2020-11-13 10:35:08,980 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.026714086532592773 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:10,684 - DEBUG - [Thread-1] UM.FileHandler.ReadFileJob.run [71]: Loading file took 3.0 seconds
2020-11-13 10:35:10,768 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.07087087631225586 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:10,821 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.039083003997802734 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:10,863 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03173685073852539 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:10,912 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03936409950256348 seconds. 164 input vertices. 164 output vertices.
2020-11-13 10:35:10,961 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.040574073791503906 seconds. 164 input vertices. 164 output vertices.
[... 36 lines cut ...]
2020-11-13 10:35:14,111 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [159]: Need to check for errors again. Discard the current progress and reschedule a check.
2020-11-13 10:35:14,453 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2020-11-13 10:35:14,484 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2020-11-13 10:35:14,506 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura 4.8.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
2020-11-13 10:35:14,558 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-11-13 10:35:15,386 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.8s
2020-11-13 10:35:15,401 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura 4.8.app/Contents/MacOS/resources/qml/ObjectItemButton.qml:81:9: QML Button: Binding loop detected f [... cut ...]
2020-11-13 10:35:18,798 - WARNING - [Thread-2] CuraEngineBackend.StartSliceJob.run [224]: No objects suitable for one at a time found, or no correct order found

**Additional information**
Printing "one at a time" helps me print lots of mask straps.  If one strap goes bad or slips slightly its less likely to affect other straps being printed than a layer slipping and causing the whole bed to be ruined.  The straps are only 2 x .8mm layers tall and no risk of hitting the print head/shroud let alone the gantry.

GregValiant, 2020-11-13:
In machine settings for your printer, on the right side of the dialog are the dimensions of the print head.  Those dimensions define the non-printable area due to interference that can occur with another part on the build plate.  They come into play when you select "one at a time".  Cura will not slice if it appears there will be contact between the print head and a finished part.  With just two parts on the build plate, select "one at a time" and look at the gray area around the models.  That is where the print head can crash.  Your interference areas all overlap so Cura refuses to slice.  If you spread them out so that no model touches another model's gray area it should slice UNLESS they are taller than the "gantry height".  If you really need them that close together then you need to print them all at once.

konskarm, 2020-11-16:
@GregValiant explained it perfectly. One-at-a-time mode needs space to ensure that the print head won't crash on one of the object that is already printed on the buildplate. This is expected behavior so I am closing this.

closed 2020-11-16 by konskarm