Ultimaker/Cura #8763: <!--I am resubmitting this issue #8413

truegret, 2020-11-13:
Cura 4.7.1 MacOS - no mountable file system [El Capitan 10.11.6]

The logs are with the ticket plus screen shot.  Please read the whole ticket before asking for more logs and/or provide specific steps to retrieve additional logs.

**Application version**
Ultimaker_Cura-4.7.1-Darwin

**Platform**
MacOS 10.11.6 - El Capitan
Security & Privacy | Allow apps downloaded from: Anywhere

**Printer**
Creality CR10S

**Reproduction steps**
First attempt
1. In response to "New version available" upon launching of Cura, downloaded 4.7.1 for MacOS
2. Navigated to file location
3. Double-click on Ultimaker_Cura-4.7.1-Darwin.dmg
4. Window to unpack application started
5. Warning window popped up "The following disk image couldn't be opened"
    5.1 Image: "Ultimaker_Cura-4.7.1-Darwin.dmg" Reason: "no mountable file systems"

Second attempt
0. Read release notes 
1. Download Cura 4.7.1 within Chrome browser (https://ultimaker.com/software/ultimaker-cura)
    1.1 Downloaded from Safari - same error
2. Navigated to file location
3. Within System Preferences set Security & Privacy settings
    3.1 Allow apps downloaded from:
    3.2 FROM: Mac App Store and identified developers
    3.3 TO: Anywhere
4. Double-clicked on Ultimaker_Cura-4.7.1-Darwin.dmg
5. Window to unpack application started
6. Warning window popped up "The following disk image couldn't be opened"
    6.1 Image: "Ultimaker_Cura-4.7.1-Darwin.dmg" Reason: "no mountable file systems"

Third attempt
Same results as Second attempt when downloaded using Safari 

**Screenshot(s)**
See attached
Performed in Terminal:
[dgretlein-imac:Ultimaker Cura dgretlein$ hdiutil attach -verbose Ultimaker_Cura-4.7.1-Darwin.dmg 

**Actual results**
1. Window to unpack application started
2. Warning window popped up "The following disk image couldn't be opened"
    2.1 Image: "Ultimaker_Cura-4.7.1-Darwin.dmg" Reason: "no mountable file systems"

**Expected results**
Dialog window allowing me to select where the application is dropped into, and/or with the Applications folder

**Project file**
NA

**Log file**
See attached screenshot

**Additional information**
I successfully launched Ultimaker_Cura-4.7.0
[image]

__Originally posted by @truegret in https://github.com/Ultimaker/Cura/issues/8413__

nallath, 2020-11-16:
Well, a re-submit is a duplicate, so I will close this in favor of 8413

closed 2020-11-16 by nallath

truegret, 2020-11-16:
Hi

Has the issue of starting Ultimaker Cura 4.7.1 on El Capitan been resolved?

David

iP7 GMail email
iOS 14.1 (18A8395)

Ghostkeeper, 2020-11-17:
Ah, yes I believe it was solved in our process for 4.8.0. Should be working both in beta and in stable.