Ultimaker/Cura #8764: Cura 4.8.0 stabil can not find octoprint instances

00monter00, 2020-11-13:
Thank you for making Cura ;)

**Application version**
4.8.0 stabile

**Platform**
win10 64 pro

**Printer**
ender 3 pro

**Reproduction steps**
1. update all apps also ostoprint connection to 3.5.1.7

**Screenshot(s)**
https://i.postimg.cc/m26wLZKZ/octo.jpg

**Actual results**

**Expected results**

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I have not any issue with 4.8.0 beta, 4.7.1 and 4.6 (still i have). I'm not only one with this issue.

fieldOfView, 2020-11-13:
Issues with the OctoPrint plugin should be reported in the issue queue for that plugin:
https://github.com/fieldofview/cura-octoprintplugin/issues

If you report an issue there, be sure to also include what version of OctoPrint you use.
If you are running OctoPrint 1.5.0 RC1 on Python 2, see this issue for OctoPrint: https://github.com/OctoPrint/OctoPrint/issues/3813

jellespijker, 2020-11-13:
Thank you for referring him @fieldOfView
I asked if he could report this on FB but I wasn't clear in my communication and forgot to mention that it should be reported in the octoprintplugin issue queue. 

@00monter00 I will close this issue here but thanks for taking the time to provide us developers feedback.

closed 2020-11-13 by jellespijker