Ultimaker/Cura #8765: Model merge and layout issues in 4.8.0

hexxxx, 2020-11-14:
**Application version**
4.8.0

**Platform**
Win 10 Enterprise

**Printer**
UM S5

**Reproduction steps**
1. Import two STLs 
2. Assign one STL to Extruder 2
3. Select both models 
4. Right click -> Merge Models

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
Position of merged model is way, way off. It's off the build platform, rotating the model spins the merged parts around in a circle with a radium seemingly larger than the build volume, and no amount of manual adjustment can make Cura recognize the print as being within bounds. 

**Expected results**
Models merge appropriately, experience no exaggeration of size, and are printable.

**Project file**
[bug report project file.zip](https://github.com/Ultimaker/Cura/files/5539935/bug.report.project.file.zip)

**Log file**
[log excerpt.txt](https://github.com/Ultimaker/Cura/files/5539964/bug.log.excerpt.txt)

**Additional information**
This was happening to some extent in the previous version of Cura, but this is much more dramatic. I posted this previously, but let the issue close because I got tied up with work and neglected to respond. In prior Cura versions the issue was manageable, and could be worked around.

nallath, 2020-11-16:
Could you also share the STL files to reproduce this (This is the only case where we actually need the STL's instead of the project ;))

Ghostkeeper, 2020-11-17:
I could reproduce this issue by exporting the models from that project file to STL and then loading those back in. Here are STLs that reproduce the problem: [positioning_bug.zip](https://github.com/Ultimaker/Cura/files/5556416/positioning_bug.zip)

hexxxx, 2020-11-17:
Original STLs [here](https://www.dropbox.com/s/lno34g6ol1wdf9m/positioning_bug_original_stls.zip?dl=0)! Sorry for the delay.

Vandrasc, 2020-11-23:
Devs see CURA-7873 for our own internal reference.

hexxxx, 2020-11-26:
Worth noting that I'm having the same issues with different STLs of similar type and complexity, while attempting the same merge operations.

Ghostkeeper, 2021-05-21:
This happened when a model is transformed (scaled or rotated) before merging. The reason it started happening more in Cura 4.8 is because the new arrange algorithm automatically rotated models when loading them in. The bug will be fixed in #9884, set for merger in 4.10. As a workaround, you can reset all transformations before merging.

PR #9884 (2021-06-08): Fix positioning of merged transformed meshes
The other transformations, in particular rotation, were also influencing the merge. The transformations were not removed when calculating the center position, causing the center position of a mesh to be calculating differently. When the meshes were actually transformed/moved, all of the transformations were removed but this gave the model a different center position. The different center position wasn't properly being compensated for using the previously calculated center position.

This caused a bug to appear where merged meshes were wildly reposit [... cut]
closed 2021-06-08 by rburema