Ultimaker/Cura #8766: Flow options for raft top surface

viper0078, 2020-11-14:
I make rafts with ABS filament, but the rafts often stick to the printed material and are difficult to remove.

By increasing the flow rate of the raft top surface, the strength of the raft is increased and the raft does not adhere to the printed material.
On the right is the raft with a 20% increase in flow rate.
[image]
[image]

I would like to see this option added.
Please consider it.

Timocop, 2020-11-15:
I think you just need to tune the raft still. By looking at your pictures seems like you are pushing too much on the raft, making the model stick too well.
The problem with Cura's raft is that most of it is hard-coded. Means, if you use a different nozzle than 0.4mm you will run into problems. Either sticking too well, or never stick at all. I've spend months finding the perfect universal raft air-offest math using different nozzles (0.3mm, 0.4mm and 0.6mm) and have successfully printed ABS even without top layers now. Here it is what i call 'universal air-offset':
**Initial Layer Height:  (Line Width / 2)
Raft Air Gap: (Line Width / 2)
Initial Layer Line Width: 100%**

viper0078, 2020-11-15:
My printer's nozzle size is 0.4mm, and the settings are as follows

Initial Layer Height: 0.2mm
Raft Air Gap: 0.13mm
Initial Layer Line Width:100%

It is best because it is easy to peel off from the raft and does not warp.
The filament is Pxmalion ABS.

The optimal value for RaftAirGap when I use PLA is 0.15mm.

If you set Raft AirGap to LineWidth/2 (0.2mm), the ABS will have a weak fixation and flake and warp the printout.
Wouldn't the adjustments be different for different printer models and filament types?

My old printer used to adjust the level of the bed with spring screws. Sometimes, though, there was a slight shift in the level and the raft would stick badly to the printed material. So I figured out a way to increase the strength of the raft so that it would be easier to peel it off the printed material. Since then, I've been using C# to create a program to convert the GCODE.
My current printer is a SnapMaker, though, so the bed level is unchanged.
I use the conversion program to prevent warping of the ABS and welding of the rafts.
I posted this because I wondered if some of you might have the same difficulty.

viper0078, 2020-11-15:
I also make another conversion process to clean up the printouts.
If the raft is difficult to remove from the printout, some of the first layer of the printout may peel off and become dirty.
To prevent this, I increased the flow rate of the second layer, except for Shell, a little .
The second layer adheres strongly to the first layer, so the first layer doesn't partially peel off when the raft is removed.

Probably won't be adopted. Haha.

Timocop, 2020-11-16:
My printer is a cheap Geeetech i3 Prusa Pro W and the filament is also pretty cheap too haha. Normally ABS sticks to itself very well. Never had any problems with my universal air-gap. Mostly because i was finding a sweat spot between ABS good sticking and PLA less sticking. PLA on raft is a pain to remove, mostly breaking the raft in the process. If i wanted to make the bottom finish nicer, i'd increase "Initial Layer Line Width" a little so the filament doesn't touch other lines making them bend.

I surely hope they implement raft flow rates, not just for the top layers, also the bottom layers which are way more important.
Already opened an issue on this: https://github.com/Ultimaker/Cura/issues/8525

Timocop, 2020-11-16:
The current workaround for you is to increase the ``Raft Top Line Width``, but keeping the ``Raft Top Spacing`` the same.

Ghostkeeper, 2020-11-17:
The "line width" settings effectively function as flow settings for the raft, indeed. It's one of those things that has been weird since Cura's conception but we never really get around to changing.

bishop-5, 2020-11-19:
I can achieve the same function with line width and line spacing. I didn't notice it.
Thanks for the answer.

viper0078, 2020-11-19:
Sorry, I commented on a different computer, so my ID was bishop-5.