Ultimaker/Cura #8767: Preview mode in Cure not working on Mac OS 11 Big Sur

SirMest, 2020-11-14:
**Application version** 4.8
(The version of the application this issue occurs with.)

**Platform** Mac OS 11 Big Sur
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer** Ender 3
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.) Any job work, you just can't see it on the Preview screen

2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) [image]

**Actual results**
(What happens after the above steps have been followed.) blank screen

**Expected results**
(What should happen after the above steps have been followed.) Preview of the model should be seen

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

dansharkey, 2020-11-15:
I had this issue, but it looks like a software restart has fixed it. Can you confirm the issue is still present after restarting Cura? If you can search the log file (https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload), that would be very helpful! :)

squireblackdens, 2020-11-15:
I have same issue. Attached is log file for following process - 
1) Open app
2) Load model from thingverse plugin
3) Slice
4) Preview

As OP, blank screen on preview. Macbook Pro, Big Sur installed this morning (which I'm beginning to regret!0
I have only just installed Cura (getting my head around it before printer arrives) so can't say if I had this problem before Big Sur.
Thanks!
[cura.log](https://github.com/Ultimaker/Cura/files/5542781/cura.log)

fieldOfView, 2020-11-15:
Could you try checking the "Force layer view compatibility mode (restart required)" and restarting Cura?

squireblackdens, 2020-11-15:
HI. Thanks for the quick response. Tried "Force layer view compatibility mode" - preview still blank and now play controls disappear from bottom. Come back if I turn it off. Thanks.

fieldOfView, 2020-11-15:
The part in your logs that corroborates your issue is this:
```
2020-11-15 17:04:54,588 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/SimulationView/simulationview_composite.shader]...
2020-11-15 17:04:54,594 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setVertexShader [121]: Vertex shader failed to compile: 
2020-11-15 17:04:54,597 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setFragmentShader [132]: Fragment shader failed to compile: 
2020-11-15 17:04:54,600 - ERROR - [MainThread] UM.View.GL.ShaderProgram.build [148]: Shader failed to link: 

```
A shader does not compile. Unfortunately, the actual errors are not being logged.

squireblackdens, 2020-11-15:
Thanks. do you think this is a Big Sur / GL issue or something else? Sorry to be so lacking in knowledge!

fieldOfView, 2020-11-15:
It is hard to tell what is going on because the actual error messages are missing.

squireblackdens, 2020-11-15:
OK Thanks. I was only using it to get familiar before my printer arrives so I'll spin up a linux VM on my unraid server and try it on their. My guess is it's Big Sur related. Along with other issues I'm having may restore to catalina tomorrow.

SirMest, 2020-11-15:
Hi. Took a couple of restarts, but all seems to be working as it should again. Have had this issue with some other software since updating to Big Sur. Have had to uninstall and reinstall to the same version to get some other software working again.

closed 2020-11-15 by SirMest

deftdawg, 2020-11-17:
I have this problem too, I uninstalled 4.8.0 and went back to 4.7.1 and no problems.   Running Big Sur, didn't need to reboot after downgrading to 4.7.1.

akarpenko, 2020-11-29:
Reverting to Defaults and then restarting Cura helped me get the preview in 4.8.0 on Big Sur.

dmelgar, 2020-12-31:
Any news on a fix that doesn't require reverting Mac OS, or Cura, or resetting settings? I don't want to lose my custom configurations for materials etc. I'd rather not revert Mac OS nor Cura.