Ultimaker/Cura #8769: unnecessary tool changes when printing with one head only

AndreasALu, 2020-11-14:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
CraftBot Flow IDEX

**Reproduction steps**
I´m printing multiple opbjects "one at a time" with head #1. Head #2 is disabled in the printer settings. At the start of the G-code and at every new object, there is a T0 command, which causes the printer to move the head to the park position and on the way, it smashes into already printed objects. 

Also, the printer´s firmware does an automated retract / prime when the tool changes, which would be ok if the tool actually changes but causes problems when no tool change.

When the T0 command is taken out (e.g. SearchAndReplace postprocessing), all goes as expected. 

When the number of extruders is set to 1 in the printer definitions, no T0 command is given. Which is correct.

If the number of extruders is 2 and the second extruder is disabled, the T0 is given. Which is not correct and causes problems on the printer (and possibly other printer models as well).

**Actual results**
T0 command is sent even if no tool change happens.

**Expected results**
No tool change command if tool does not change or tools are disabled and only 1 tool is activated.

rburema, 2020-11-17:
In the machine settings, what is the gcode-flavour set to? (Also, what are the machine- and extruder-start and end-gcodes?)

Ghostkeeper, 2020-11-17:
A project file (one of the entries from the issue template that you egregiously removed) would help to reproduce the issue.

AndreasALu, 2020-11-26:
> In the machine settings, what is the gcode-flavour set to? (Also, what are the machine- and extruder-start and end-gcodes?)

G-code flavour is Marlin. No Extruder Start- / End- G-code.

Machine Start G-code:

M9006 S0 ;select normal Dual Extruder Mode
M104 T0 S{material_print_temperature_layer_0, 0}
M140 S{material_bed_temperature_layer_0} 
G28 X Y ;home X and Y axes
G28 Z ;home Z
M190 S{material_bed_temperature_layer_0} 
M109 T0 S{material_print_temperature_layer_0, 0} 
M425 S1 ;use wiper blade
G21 ;mm
G90 ;abs
G92 E0
G1 Y140 F1800 E10

----------------------------------------------------
End:
G101 Z10 ;lower the bed 10mm
G90
G28 X Y ;home X and Y
G0 Y250 ;move Y back away from the camera

M104 T0 S0;turn off head 1 (left)
M140 S0 ;turn off bed
M106 T0 S0 ;turn off fan on head 1 (left)
M84 ;disable motors

AndreasALu, 2020-11-26:
Here´s a sample project and the G-code. Note the T0 after every object.

[Three Cubes One At A Time with T0 Tool Change.zip](https://github.com/Ultimaker/Cura/files/5602983/Three.Cubes.One.At.A.Time.with.T0.Tool.Change.zip)

Ghostkeeper, 2020-11-27:
Okay, so this is coming from the `processInitialLayerTemperature` function which also outputs the `T0` at the start: https://github.com/Ultimaker/CuraEngine/blob/0852f3cd5e909c9ebb459befb499292460da97d4/src/FffGcodeWriter.cpp#L492

The `T0` at the start is a necessity because this state before the start of the print is unknown. However this is not necessary for each subsequent print in one-at-a-time mode. Arguably this also shouldn't be part of a function with the name `processInitialLayerTemperature` since it's got little to do with the temperatures.

Ghostkeeper, 2020-11-30:
(Developers, refer to issue CURA-7895.)

3Duke3, 2022-02-04:
> Okay, so this is coming from the `processInitialLayerTemperature` function which also outputs the `T0` at the start: [...]

Sorry to hijack this thread, but it seemed similar. Is it possible for an option to remove the very first T0 generated at the beginning of gcode? Or for example, if T-1 (or even T0?) is manually added in my start gcode, then the first T0 is removed? 

I understand why the T0 needs to be there, but in some cases we run functions like purge, wipe, retract ect in the tpre and tfree macros (Duet), so its of mild annoyance to watch T0 be picked up, try to run a bunch or stuff, only to be put back and then start its heating process and do it all over again.

closed 2023-06-24 by github-actions[bot]