Ultimaker/Cura #8770: Bridging line directions should be an option without enabling bridge settings.

printingotb, 2020-11-14:
**Is your feature request related to a problem? Please describe.**
I want to be able to use the bridge line direction without enabling bridge settings.
Making horizontal overhangs and the first layer on infill bridge better.

**Describe the solution you'd like**
Add a check-box under shell settings called "Optimize line direction for overhangs" or something like that.

**Describe alternatives you've considered**
Enabling bridge settings, and setting them all to behave as if bridge settings are not active.
This is somewhat inconvenient since I would need to change the bridge fan speed from case to case. Saving the setting with a low value would reduce the regular fan speed if it is higher and saving it with a high value can ruin prints with high temp materials.

I would also have to save either my best bridging settings or the bridge settings effectively disabled and change the bridge fan speed from case to case.

I hope this makes sense, I want to be able to create highly optimized baseline profiles to share with others without them needing to change and understand things like this. And without using workarounds.

**Affected users and/or printers**
Everyone

**Additional context**
[image]

Ghostkeeper, 2020-11-17:
Cura does detect bridges and adjust the line directions accordingly even if "Enable Bridge Settings" is disabled. Here is an example of a simple print where it adjusts the line directions: [UM3_bridge.zip](https://github.com/Ultimaker/Cura/files/5556518/UM3_bridge.zip)

But there is a difference: the Bridge Skin Support Threshold setting. Because of this setting, the detection algorithm for bridges is completely different if the bridge settings are enabled. In my opinion, when the PR for those bridge settings was merged, we should also have removed the old bridge detection algorithm and just used a default threshold of 50% or something. But we didn't and we're sort of dealing with the maintenance of two algorithms that do effectively the same now, but do have a slightly different result. I would currently call this expected behaviour, but it touches the issue of whether we should merge the two algorithms.

printingotb, 2020-11-18:
Hmm, it does, how did I miss that. I can't get it to do it on top of infill though? Even when set to 0%. Can that be done?

Does it calculate the direction based on a "centerline" (in lack of a correct term) of a an area? As in the area is wider in one end and therefore the line direction isn't parallel to the bridge.
[image]

Noticed this snow. Only difference is that the lower part's leg is thinner and the skin doesn't have any room to expand into it.
[image]

Ghostkeeper, 2020-11-19:
If I recall correctly, if the bridge settings are disabled, it will find the angle between the centres of mass of the two parts that the bridge rests on. This is a bit of a weird metric I think, and it only works if it's actually bridging between two distinct parts (which is a common case, but still).

If bridge settings are enabled, it takes the direction of the longest unsupported side, which aligns well with the shape of the model if the shape is straight and mechanical like your case. It doesn't work very well with organic shapes then, but those don't often have large bridges anyway.

Orum, 2021-01-13:
Is there a way to disable the automatic bridging line direction?  For some parts it makes _very_ long bridges:
[image]

Rotating the part, toggling bridging settings on/off, and even adjusting "Bridge Skin Support Threshold" has no effect on how the bridging is done.

fvrmr, 2021-01-19:
@Orum It is not possible to disable the automatic bridging line direction. Like @Ghostkeeper mentioned above:

> If bridge settings are enabled, it takes the direction of the longest unsupported side, which aligns well with the shape of the model if the shape is straight and mechanical like your case.

Orum, 2021-01-20:
> If bridge settings are enabled, it takes the direction of the longest unsupported side

The issue is that this occurs even with bridge settings completely disabled.  There is no way to get it to do a 45-degree bridge on some models, making it impossible to slice certain models with Cura.  I gave up and used PrusaSlicer for the model above, but it'd be nice to be able to slice it in a printable manner with Cura.

Ghostkeeper, 2021-01-25:
> The issue is that this occurs even with bridge settings completely disabled.

The extra bridge settings are disabled, but bridging itself is not. To customise where it's going to be bridging, you need to *Enable* the extra bridge settings. You can then set the Bridge Skin Support Threshold to 0% to effectively disable bridging altogether.

Orum, 2021-01-25:
I've done that, it doesn't change the result:
[image]

Ghostkeeper, 2021-01-27:
Perhaps a project file would allow us to reproduce that, then.

Orum, 2021-01-27:
Unfortunately, Cura crashes whenever I try to save a .3mf (I've submitted the crash report via the internal crash reporter).  However, I've attached [the STL](https://github.com/Ultimaker/Cura/files/5881951/example.zip), and will include the basic settings to repeat the issue:

- Printer: Ender 3
- Supports: disabled
- Layer Height: 0.16mm (problem is visible on layer 11; go to layer 10 to see what it's bridging over)
- Bridge Settings: Tried both enabled (with 0% skin support threshold) and disabled, with no change.
- Other things tried: Rotating part around the Z-axis before slicing, with no change.

Let me know if you need anything else to reproduce the issue.

Ghostkeeper, 2021-01-28:
Yeah I think that reproduces the issue.

When investigating, I found an edge case that wasn't being handled well. I fixed it here: https://github.com/Ultimaker/CuraEngine/pull/1399

With that edge case fixed, it's no longer bridging there.
[image]

As a workaround, try setting the Bridge Skin Support Threshold to 0.1% instead. In my case at least that makes it not bridge any more, as desired. The special handling of the 0% case was skipping over some code and assuming it was bridging, which was faulty.