Ultimaker/Cura #8773: "Open Project" Dialog too small to show printer names

KenLucke, 2020-11-14:
**Application version**
4.8

**Platform**
MacBook Pro/Mac OS 10.15.7 (no, I'm not ready to upgrade to 11.0 yet, until bugs are worked out in the OS)

**Printer**
N/A

**Reproduction steps**
1. Open a  saved project
2. Attempt to set printer profile to update

**Screenshot(s)**
[image]

**Actual results**
Dropdown is too narrow to show some full printer names, so it's a guessing game as to which one to choose when similar printer names exist.

**Expected results**
Dropdown control should show full names, or at least the tooltips when hovering should

**Project file**
N/A, happens in all projects

**Log file**
N/A

**Additional information**
N/A

Ghostkeeper, 2020-11-17:
Thanks for the report! I've fixed it for 4.9.

Ghostkeeper, 2020-11-17:
Until then, you can make the dialogue wider to display more text. This combobox scales with the total width of the dialogue.

commit 558a7892 (2020-11-17): Use Cura's own ComboBox element for combo boxes

This one displays a tooltip if the text is too long and gets elided, so that you can still read the printer or profile name.

Fixes #8773.
closed 2020-11-17 by Ghostkeeper