Ultimaker/Cura #8774: Using "Add material from Marketplace" says it installed the material (e.g. Ninjatek ...) but the addition does not show up on the material select list.

shilo91, 2020-11-14:
"Add material from Marketplace" says it installed the material (e.g. Ninjatek ...) but it does not show up on the material select list after restarting Cura.

**Application version**
Happens on both 4.7.1 and 4.8.0 versions.

**Platform**
Windows 10

**Printer**
Creality Ender 3 and others.

**Reproduction steps**
1. Opened add material screen, selected Ninjatek and pressed install. Install completed and Ninjatek now shown with checkmark.
2. Tried to select Ninjatek from material list - no listing for Ninjatek appears.

**Screenshot(s)**
No point!

**Actual results**
User is annoyed!

**Expected results**
Able to select newly added material.

**Project file**

**Log file**
No log file available as was unable to selelct material.

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-11-16:
That's because the material that you installed is a 2.85 material, not a 1.75 material.

closed 2020-11-16 by nallath

shilo91, 2020-11-16:
Bullshit! – Ninjatek Armadillo filament is available in both 1.75 & 3 mm diameters as are most other filaments. The listing on the Marketplace page in Cura does not indicate the diameter of the filament. If you click on Buy Material Spools you are directed to a web site order screen where the diameter is selectable. Your statement that I installed a 2.85 diameter material is wrong as I was given no means to specify the filament size nor was the size indicated on the Marketplace screen.

Now back to the original issue - the addition does not show up on the material select list.  I believe the correct fix to this problem is to update the Cura database for all available filament diameters
for the material being added. If that is too difficult then at least specify the filament diameter on the Marketplace listing.

From: Jaime van Kessel
Sent: Monday, November 16, 2020 2:35 AM
To: Ultimaker/Cura
Cc: shilo91; Author
Subject: Re: [Ultimaker/Cura] Using "Add material from Marketplace" says itinstalled the material (e.g. Ninjatek ...) but the addition does not show upon the material select list. (#8774)

That's because the material that you installed is a 2.85 material, not a 1.75 material.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

fieldOfView, 2020-11-16:
Perhaps @nallath should have said "That's because the material_profile_ that you installed is a 2.85 material, not a 1.75 material."

All material profiles currently in the Marketplace only include data for 2.85 mm variants. Calling that bullshit does not change that. These material profiles are intended for Ultimaker printers, and all Ultimaker printers use 2.85 mm materials. There is no data for your printer in the material files in the Marketplace, and that is why the material does not show up. This is not the best UX for users of non-Ultimaker printers, but it is how it is.

shilo91, 2020-11-16:
Perhaps this is something that should be made more apparent for non-Ultimaker printer users of Cura – Such as a line on the Marketplace screen to indicate that only 2.85 materials are listed
and that items listed are specific to Ultimaker brand printers.

From: Aldo Hoeben
Sent: Monday, November 16, 2020 12:15 PM
To: Ultimaker/Cura
Cc: shilo91; Author
Subject: Re: [Ultimaker/Cura] Using "Add material from Marketplace" says itinstalled the material (e.g. Ninjatek ...) but the addition does not show upon the material select list. (#8774)

Perhaps @nallath should have said "That's because the material_profile_ that you installed is a 2.85 material, not a 1.75 material."
All material profiles currently in the Marketplace only include data for 2.85 mm variants. Calling that bullshit does not change that. These material profiles are intended for Ultimaker printers, and all Ultimaker printers use 2.85 mm materials. There is no data for your printer in the material files in the Marketplace, and that is why the material does not show up. This is not the best UX for users of non-Ultimaker printers, but it is how it is.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Ghostkeeper, 2020-11-17:
We have plans - quite short term - to make it more clear in the marketplace which configurations are supported by the material profile in the Marketplace. It's sometimes a problem within the scope of Ultimaker printers as well.

shilo91, 2020-11-18:
Restarted CURA (4.8.0) and temporarily modified the machine settings for one printer from 1.75 to 2.85 and verified the material appeared on the select list, changed it back to 1.75 and verified the new material was not listed.

Shut down the computer and on start-up the next day low and behold the recently added Ninjatek material is now included in the list of 1.75 materials! Magic??
From: Ghostkeeper
Sent: Tuesday, November 17, 2020 3:51 PM
To: Ultimaker/Cura
Cc: shilo91; Author
Subject: Re: [Ultimaker/Cura] Using "Add material from Marketplace" says itinstalled the material (e.g. Ninjatek ...) but the addition does not show upon the material select list. (#8774)

We have plans - quite short term - to make it more clear in the marketplace which configurations are supported by the material profile in the Marketplace. It's sometimes a problem within the scope of Ultimaker printers as well.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

shilo91, 2020-11-18:
Update on Material lists: I currently have 6 printers defined in  Cura all defined as using 1.75 filament, Printers have been added as new printers were purchased or merely to evaluate a particular model. For some printers the material select list is Generic+18 Manufacturer Specific while for other printer the list generated is Generic + 8 Manufacturer Specific. I just added a Creality CR10 to see what a newly added 1.75 printer would get - Generic+8 Manufacturer Specific. This makes little sense to me as to why some 1.75 printers would get a different material list. It also makes no sense that filament size would be considered in the Cura Material list - manufacturers seem to provide only one Printing Guideline spec for each product regardless of the diameter. I would think the extruder feed rate calculations would handle the difference in filament diameter.

Ghostkeeper, 2020-11-19:
Some printers have explicit mentions of materials that are disallowed, such as this HMS434 which lists a whole bunch of them: https://github.com/Ultimaker/Cura/blob/39acad87d916b32c98e852b9e14b6b76b0070852/resources/definitions/hms434.def.json#L12-L31

If the filament diameter is different, the heating of the filament happens at a different speed. After all there is less surface area on the filament for the heat to pass through. The middle takes longer to melt. Because the heating is different, the viscosity of the plastic is different and then the whole printing process is different. Also retractions have more effect with wider filament. Wider filament causes less delay when the feeder is far from the nozzle (because it's stiffer). And so on. Not all filaments work the same way with 2.85mm as with 1.75mm, so just having the extruder move more slowly is not a solution.

shilo91, 2020-11-20:
Did not know there were exclusion parameters in printer profiles – never came across references to this on any blog or tutorial I’d read. 
I recognized the effect a larger diameter filament would have a different heating profile for the hotend, but considered this to be a printer specific issue and not directly
related to the material. But back to the original point – Marketplace should be more descriptive if an entry is meant for only a specific printer or type of printer. Cura is
no longer a tool for only Ultimaker printers as a version is included with most, if not all, printers you buy. New users should be able to use the feature without the hassle of
dealing with vague item descriptions. 

From: Ghostkeeper
Sent: Thursday, November 19, 2020 4:49 AM
To: Ultimaker/Cura
Cc: shilo91; Author
Subject: Re: [Ultimaker/Cura] Using "Add material from Marketplace" says itinstalled the material (e.g. Ninjatek ...) but the addition does not show upon the material select list. (#8774)

Some printers have explicit mentions of materials that are disallowed, such as this HMS434 which lists a whole bunch of them: https://github.com/Ultimaker/Cura/blob/39acad87d916b32c98e852b9e14b6b76b0070852/resources/definitions/hms434.def.json#L12-L31
If the filament diameter is different, the heating of the filament happens at a different speed. After all there is less surface area on the filament for the heat to pass through. The middle takes longer to melt. Because the heating is different, the viscosity of the plastic is different and then the whole printing process is different. Also retractions have more effect with wider filament. Wider filament causes less delay when the feeder is far from the nozzle (because it's stiffer). And so on. Not all filaments work the same way with 2.85mm as with 1.75mm, so just having the extruder move more slowly is not a solution.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus