Ultimaker/Cura #8775: Cura crashes on file opening

RBO85, 2020-11-15:
**Application version**
4.7.1 & 4.8.0

**Platform**
MacOS Big Sur (11.0.1)

**Printer**
Creality Ender-3

**Reproduction steps**
1. Open file via Drag-Drop, via file menu or via export function on Fusion360 (all same result)

**Screenshot(s)**
[image]

**Actual results**
App crashes instantly

**Expected results**
File should be opened

**Project file**
unable to create project file since app crashes instantly on opening file

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5542382/cura.log)

**Additional information**
N/A

nallath, 2020-11-16:
What I can see from the logs is that something seems to be corrupted with your machine configuration. 

Could you try removing the machine and re-adding it? 

If that still doesn't resolve the problem, could you share your configuration folder?

RBO85, 2020-11-17:
Hi Nallath,

Re-adding the printer did indeed solve the issue. Thanks a lot for your help.

closed 2020-11-17 by RBO85