Ultimaker/Cura #8776: Support Blocker Unable to Duplicate

esup, 2020-11-15:
**Application version**
4.8

**Platform**
Windows 10 64 bit

**Printer**
Ender 3 v2

**Reproduction steps**
1. Import model
2. Add support blocker
3. Right mouse click on support blocker
4. Multiple Model 

Nothing happens

**Actual results**
Nothing

**Expected results**
Support blocker should have been duplicated

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
2020-11-15 16:21:11,479 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=1) <UM.Operations.RemoveSceneNodeOperation.RemoveSceneNodeOperation object at 0x000001B0390D84A8>, took 4ms
2020-11-15 16:21:12,055 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=1) <UM.Operations.RemoveSceneNodeOperation.RemoveSceneNodeOperation object at 0x000001B043F60940>, took 10ms
2020-11-15 16:21:12,596 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectSelector.qml:132:23: Unable to assign [undefined] to QString
2020-11-15 16:21:12,598 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectSelector.qml:134:49: Unable to assign [undefined] to bool
2020-11-15 16:21:12,600 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectSelector.qml:135:54: Unable to assign [undefined] to int
2020-11-15 16:21:12,602 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectSelector.qml:136:43: Unable to assign [undefined] to QString
2020-11-15 16:21:12,604 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectSelector.qml:137:46: Unable to assign [undefined] to int
2020-11-15 16:21:12,893 - WARNING - [MainThread] cura.UI.PrintInformation.setBaseName [366]: Unsupported Mime Type Database file extension 
2020-11-15 16:21:14,857 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/Darren/Downloads/Octopus_v5.5_spiral.stl
2020-11-15 16:21:14,859 - INFO - [Thread-13] STLReader.STLReader.load_file [52]: Using NumPy-STL to load STL data.
2020-11-15 16:21:21,519 - DEBUG - [Thread-13] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.02299952507019043 seconds
2020-11-15 16:21:21,530 - DEBUG - [Thread-13] STLReader.STLReader._read [104]: Loaded a mesh with 907668 vertices
2020-11-15 16:21:21,754 - DEBUG - [Thread-13] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.22002649307250977 seconds. 8860 input vertices. 8860 output vertices.
2020-11-15 16:21:21,907 - DEBUG - [Thread-13] UM.FileHandler.ReadFileJob.run [71]: Loading file took 7.0 seconds
2020-11-15 16:21:22,149 - DEBUG - [Thread-13] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.23900175094604492 seconds. 8852 input vertices. 8852 output vertices.
2020-11-15 16:21:22,163 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x000001B0441192E8>, took 1ms
2020-11-15 16:21:22,181 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=2) RotateOp.(node=<CuraSceneNode object: 'Octopus_v5.5_spiral.stl'>,rot.=Q<0.0,0.0,0.0,w=4.0>) TranslateOp.(node=<CuraSceneNode object: 'Octopus_v5.5_spiral.stl'>,trans.=<0.0,0.0,0.0>), took 2ms
2020-11-15 16:21:22,385 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectItemButton.qml:81:9: QML Button: Binding loop detected for property "width"
2020-11-15 16:21:27,701 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\camera_distance.shader]...
2020-11-15 16:21:27,725 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromIndexedVertices [581]: Calculating normals took 0.0 seconds
2020-11-15 16:21:27,727 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0010018348693847656 seconds. 8 input vertices. 8 output vertices.
2020-11-15 16:21:27,732 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=2) <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x000001B044104710> SetParentOperation(node = <CuraSceneNode object: 'Eraser'>, parent_node=<CuraSceneNode object: 'Octopus_v5.5_spiral.stl'>), took 1ms
2020-11-15 16:21:27,946 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectItemButton.qml:81:9: QML Button: Binding loop detected for property "width"
2020-11-15 16:21:31,357 - DEBUG - [Thread-14] UM.Operations.OperationStack.push [72]: GroupedOp.(#=3) <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x000001B0453F7E48> RotateOp.(node=<CuraSceneNode object: 'Octopus_v5.5_spiral.stl'>,rot.=Q<0.0,4.0,0.0,w=-2.4492935982947064e-16>) TranslateOp.(node=<CuraSceneNode object: 'Octopus_v5.5_spiral.stl'>,trans.=<-73.7,0.0,-36.9>), took 3ms
2020-11-15 16:21:31,554 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-11-15 16:21:31,914 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2020-11-15 16:21:31,922 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2020-11-15 16:21:32,167 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectItemButton.qml:81:9: QML Button: Binding loop detected for property "width"
2020-11-15 16:21:32,172 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectItemButton.qml:81:9: QML Button: Binding loop detected for property "width"
2020-11-15 16:21:32,189 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.6s
2020-11-15 16:21:35,860 - WARNING - [Thread-15] cura.Arranging.Nest2DArrange.findNodePlacement [49]: Object Eraser cannot be arranged because it has no convex hull.
2020-11-15 16:21:35,883 - DEBUG - [Thread-15] UM.Operations.OperationStack.push [72]: GroupedOp.(#=0), took 0ms
2020-11-15 16:21:36,056 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-11-15 16:21:36,438 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2020-11-15 16:21:36,441 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2020-11-15 16:21:36,577 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.5s
2020-11-15 16:21:45,858 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

rburema, 2020-11-17:
Hi @esup , We think it's likely this is due to the changes to the arrange algorithm (which gets called after duplicate to place the new object). We can confirm that this is indeed a bug, and (for our reference) has been placed in our internal backlog as CURA-7851.

`2020-11-15 16:21:35,860 - WARNING - [Thread-15] cura.Arranging.Nest2DArrange.findNodePlacement [49]: Object Eraser cannot be arranged because it has no convex hull.`

erick54-soul, 2020-11-20:
I'm having the same problem,When I try to use support block in my ender 3 it just don't work and I don't know why

Vandrasc, 2020-11-23:
Devs see CURA-7851 for our own internal reference.

nalorin, 2021-06-15:
I'm a little confused - is this going to be resolved?

Both this ticket and #9096 (which outlines the same problem) have been closed and each one points to the other.

I have the same issue, and it's rather time consuming to not be able to copy erasers for cutting (especially since I have to then re-create the no-wall, no-infill, no-top/bottom settings for each eraser as well)

rburema, 2021-06-16:
@nalorin  This ticket isn't closed. The 'closed' there refers to the linked ticket. You'd see closed on  the bottom if this ticket was actually closed. Check the way it's shown in the other ticket versus this one. Can be confusing indeed if it's the last item in the issue. It has tripped me up multiple time now.

Work on internal ticket CURA-7851 hasn't started yet, it's unclear if we'll make it for next version. It is still on our backlog as a 'critical', so we'll definitely get to it at some point.