Ultimaker/Cura #8777: [Feature / plugin / script request] Intelligent fan control

Sophist-UK, 2020-11-15:
**Is your feature request related to a problem? Please describe.**
This feature request is prompted by a [CNC Kitchen video](https://www.youtube.com/watch?v=Bl2ESvtBiLo) which compared the layer adhesion strength at various fan settings.

In essence, layer adhesion is better with no cooling but surface quality, particularly on overhangs is worse.

**Describe the solution you'd like**
So it occurred to me that a feature / plugin / script could be used to vary the fan by extrusion type i.e. no fan on infill, or inner walls, but fan on full on outer-walls and top and bottom surfaces.

Because fan takes time to speed-up / slow-down, it might be necessary to pause or to insert the fan speed change earlier than the start/end of the extrusion type.

It seems to me to be likely that this could be achieved with a gCode post-processing script. But a plugin / feature with proper settings would be more user friendly.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
All users.

Sophist-UK, 2020-11-15:
P.S. My python skills are probably up to coding a gCode post processing script. So if this is considered to be the most appropriate route forward, I can probably code and submit this. 

I suspect that a solution based on a plugin / core enhancement will require changes to CuraEngine and I do not have either the skills or a development environment which I can use to do this.

Sophist-UK, 2020-11-15:
Upon reflection it seems to me that both surface quality and layer adhesion are affected by both extrusion temperature and cooling, and so it probably makes sense to have a single script / plugin to vary both of these by extrusion type.

nallath, 2020-11-16:
Well, if you can whip up a script that does this as a proof of concept, It will be a lot easier for us to convince the right people internally to invest into doing this "properly" (eg; In CuraEngine).

I think that fans react fast enough so that you don't need to worry too much about the time it takes to start.

Sophist-UK, 2020-11-16:
I have already started to write a post processing script for this. May take me a couple of weeks to write it, debug it, and do some test prints with and without it.

Ghostkeeper, 2020-11-17:
We have similar feature requests for adjusting the fan speed for overhang: https://github.com/Ultimaker/Cura/issues/6656

To a user, having a hard threshold might be simpler to understand than letting it scale along with the overhang angle.

I think also that it will be very difficult to implement anything that depends on the overhang angle in a post-processing script. To detect the overhang angle you'll need to figure out the distance in the layer below. CuraEngine does this by offsetting the shape of the layer below by a certain distance depending on the overhang angle, but in a post-processing script you don't have the shape of a layer as a polygon (just a bunch of lines), you don't have access to Clipper, and you don't have the computational power to really do such operations.

Sophist-UK, 2020-11-18:
Having fan on full for outer and inner walls would definitely help overhangs. This would be possible with this post processing script.

It might also be possible to separate inner walls into the outermost x walls and the remainder and have different fan settings for each - but this will not be included in my first version.

werdnum, 2021-06-01:
I was looking into this briefly today. Varying the fan speed by the line type (among skirt/brim, inner and outer walls, support, support interface, infill, etc) is easily possible (see [this example](https://community.ultimaker.com/topic/26361-new-post-processing-plugin-script-outerwallfan/)).

Changing the fan speed for overhang seems a bit trickier, because there's no indication in the G-Code output that a particular extrusion is for a bridge/overhang/etc.

Similarly, I think you'd have to analyze the coordinates themselves if you wanted to look for small features (maybe more than X degrees change in direction over less than Y mm).

Sophist-UK, 2021-06-01:
I will need to research what comments CuraEngine currently includes in the Gcode. If we need more comments for e.g. overhangs and bridging, then we would need CuraEngine to be enhanced. And if you are going to do that you might as well add the code to vary fan speed by type into Cura/CuraEngine.

I think that UM will need convincing that this leads to quality improvements. So a PP script that does fan adjustments for those line types it can detect might help show whether the idea is sound or not.

Ghostkeeper, 2021-06-10:
> I will need to research what comments CuraEngine currently includes in the Gcode. If we need more comments for e.g. overhangs and bridging, then we would need CuraEngine to be enhanced.

CuraEngine doesn't currently place comments for the start or end of the experimental bridging feature. The "normal" bridging is for the whole skin part so it's irrelevant then. The enhanced bridging with the experimental settings simply have their bridged lines marked as skin, outer wall and inner wall.

5axes, 2022-07-21:
> A new comment could be added to indicate the start and end of a bridged segment

Would be very nice to add more comment in The G-Code but I should certainly add a New enhancement request for that.  Idealy for me use the same "Standard" as the Prusa or SuperSlicer comment.

Actual CURA Comment  : 

;TYPE:SUPPORT-INTERFACE
;TYPE:SKIRT 
;TYPE:INNER_WALL
;TYPE:OUTER_WALL
;TYPE:INFILL
;TYPE:SKIN

Could be nice also to have 

;TYPE:BRIM   (No difference is made between a Brim and a Skirt) 
;TYPE:OVERHANG_PERIMETER
;TYPE:BRIDGE_INFILL
;TYPE:THIN_WALL

We have Also 

;TYPE:GAP_FILL

But with the 5.0 doesn't need it anymore