Ultimaker/Cura #8778: [4.8] weird bridge behaviour if the object is rotated in the buildplate

tricky2k, 2020-11-15:
Prompted by a dev in Cura's FB page, here's the bug report

**Application version**
Cura 4.8

**Platform**
Windows 10 Geforce GT710

**Printer**
Artillery X1 Sidewinder

**Reproduction steps**
Using this bridge test https://www.thingiverse.com/thing:546688, also attached here

[Bridge.zip](https://github.com/Ultimaker/Cura/files/5543307/Bridge.zip)

I noticed that changing the orientation in the buildplate has Cura to create weird patterns for the first bridge layer. If the bridge is oriented paralel to X or Y, no problem, the bridge is drawn from end to end, but if it's rotated, the shell and inner walls are drawn ok, but the bottom uses those bridge inner walls as starting points instead of drawing the whole fist layer from end to end. And in some places isn't even lines from inner wall to inner wall but mere blobs of filament, which obviously causes a mess. The color scheme for line type in Cura shows the outer lines of the bridge as Shell, the next inside as Inner walls and the rest as Bottom. Even if at engine level is detecting a bridge (I don't know), it's not showing it as such, "a bridge". Enabling Bridge settings in the Experimental Section and fiddling with the settings doesn't make a difference

Load the stl file and rotate it on the z-axis anything but 0º, 45º or 90º (it works fine then). The following screenshots are taken with the object tilted by 30º to the right

**Screenshot(s)**
[image]

[image]

**Actual results**
No pictures here, but as expected the results are awful

**Expected results**
The same object with same orientation in the buildplate sliced with another slicer. It detects the whole bridge as "bridge" by its engine, and it's shown as such using its color scheme (all red from end to end). It slices and prints as it should

[image]

**Project file**
Zipped and attached

[ASX1_Bridge.zip](https://github.com/Ultimaker/Cura/files/5543277/ASX1_Bridge.zip)

**Additional information**
Skewing the object on the plate is intended. You may have to print actual objects with bridges all over the place, some well oriented, some not, therefore you expect bridging to be consistent no matter the orientation of the object in the buildplate. Not sure if it's a bug or if the engine can't detect the bridge properly, but since if was asked to open a bug ticket, here it is

Thanks!

rburema, 2020-11-17:
Thanks for the report!

It seems there are 2 separate parts to this bug:

_The orientation_:
I'm not sure what's going on here, I'll discuss it with my colleagues.

_The infill blobs_:
I see that those strange blobs also appear to a lesser extent (when I sliced for 4.8.0) in some higher layers.

When I slice with the current main-branch version, I see no (infill) blobs, so it's must be something we fixed between 4.8.0 and the current internal builds (for the eventual next version).

I'd consider this (infill blobs) fixed for next version.

tricky2k, 2021-04-07:
The bug is still present in 4.9 beta. I think that, unless the whole first layer of the bridge is seen as bridge by Cura, not only the outer walls (I guess it's a programming thing that goes waaaay over my head), the bug will remain there

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

tricky2k, 2024-11-20:
> Is this still an issue in current Cura versions (5.8.0 and up)? Can this be closed?

Latest release, 5.9.0, and still an issue. Yep, the "Bottom" lines are straight from end to end, at least that's fixed, but it's still dropping "Infill" blobs after extruding the "Shell" and "Inner Wall". And look where it starts to drop the "Bottom", in the middle of nowhere. Bridge detection is on

[image]

GregValiant, 2024-11-20:
I'll leave this open.

I fooled around with the project file and tried to reproduce this in 5.8.0 and I haven't been able to.  All the extrusions start at the buttresses.
This is a zoom in your first posted image.  That looks like infill is trying to creep in.  I can't reproduce that.
[image]

tricky2k, 2024-11-20:
> I'll leave this open.
> 
> I fooled around with the project file and tried to reproduce this in 5.8.0 and I haven't been able to. All the extrusions start at the buttresses. This is a zoom in your first posted image. That looks like infill is trying to creep in. I can't reproduce that

It's been a while now, so I can't say for sure what profile settings did I use in 4.8.0 and 4.9 beta, but with 5.9.0, no matter what profile for Artillery X1, it does the same. Supports disabled and Bridging Settings enabled. Changing the Z-Seam to "Sharpest Corner" or "Shortest" fixes the "Bottom" starting anywhere thing, but the infill blobs are still there

GregValiant, 2024-11-20:
I printed the model on my Ender 3 Pro (it's a rainy day here) and I didn't get any blobs.  My printer sucks at bridging though and the two longest ones didn't turn out very well.  I didn't see any problem in the toolpaths (no hesitations).
Those "infill" marks are odd.