Ultimaker/Cura #8779: Object Overlap using Multiply object in version 4.8.0

tjones99, 2020-11-16:


tjones99, 2020-11-16:
Hmmm... I filled everything in... why am I seeing nothing

nallath, 2020-11-16:
We've seen this issue before, but we couldn't get reliable reproduction steps. Thanks for submitting it!

tjones99, 2020-11-16:
If there is anything else you need let me know. If it is something to do with my STL file you are welcome to that as well. It’s just a simple Tinkercad file and nothing proprietary.

Ghostkeeper, 2020-11-17:
Hmm, well I think I got it to reliably reproduce. But there's a couple of things to it:
* "Ensure models are kept apart" needs to be disabled. If this is enabled, the move-apart fixes the problem.
* It only seems to occur in certain build volumes. In my UM3 I can't reproduce the problem, but with the printer from this project file the problem occurs pretty much every time.
* The problem is related to the rotations of the model. It only happens with models that don't have a square bounding box (like the rectangle in the project file). I was able to reproduce the issue every time by simply loading a cube and scaling it non-uniformly.

tjones99, 2020-11-18:
Hmmm... "ensure models are kept apart" eh... Never noticed that option... will enable that... 
Maybe the issue is that the state of that option is by default disabled in 4.8.0??? I have never had to enable that option in previous versions, and I have used MANY previous versions, and I have use the multiply object often and I have never had overlapping models until 4.8.0. I have had cases where I tried to load more than would fit on the build area and they get the crosshatched pattern and stay outside the build area but they still don't overlap. 

Thanks for pointing out the ensure models are kept apart... I feel like a noob...

Ghostkeeper, 2020-11-19:
Even if the setting is disabled, the multiply feature is expected to place objects besides each other.

Ghostkeeper, 2021-04-16:
Developers, see issue CURA-7857.