Ultimaker/Cura #8781: The middle of the object is filled with skin?

bishop-5, 2020-11-16:
The middle of the object is also filled with skins and infills.
Is this a bug? Or does it depend on your settings?
[image]

[CFFFP_test.zip](https://github.com/Ultimaker/Cura/files/5546045/CFFFP_test.zip)

eskeyaar, 2020-11-16:
You have 3 walls, reduce to 2 see your infill.
[image]

bishop-5, 2020-11-16:
I found this in a printout I was working on.
So I created a test object to upload to the board.
I can't reduce the walls to keep the printout strong.
Is there another solution?

eskeyaar, 2020-11-16:
You have small model part, so when printing with 3 walls, it automatically takes as single line. Change ur skin overlap percentage to low value, then you can get single line inside the wall.  or   reduce wall to 2, and make infill to 100%, to make strong parts.

konskarm, 2020-11-16:
This is caused by the "Fill Gaps Between Walls", which if it is set to "Everywhere" may result in overlapping infill and skin. Duplicate of https://github.com/Ultimaker/Cura/issues/8724.

closed 2020-11-16 by konskarm