Ultimaker/Cura #8783: [4.8.0] Infill not complete

TotesTanner, 2020-11-17:
[4.8.0]
MacOS Big Sur
Creality CR10
Steps:
1.Import model
2 Scaled and rearranged model 
3.Sliced
4.Noted that infill in sliced model has defects

Link to project files and screenshots:
https://drive.google.com/drive/folders/1GJ4Ooo6WlU6jsTYjNRmvTFVmkshim9ue?usp=sharing

rburema, 2020-11-17:
I see you have 'Infill Support' on, which states (in the tooltip) 'Print infill structures only where tops of the model should be supported. Enabling this reduces print time and material usage, but _leads to ununiform object strength_' (emphasis mine). Indeed, when I turn off 'Infill Support', the behaviour goes away.

I'll discuss if this is what we're expecting from that setting with my colleagues anyway, as it seems somewhat counter-intuitive even if it may work as expected.

TotesTanner, 2020-11-17:
Perfect! so glad you were able to figure it out! I must have flipped that switch thinking it was simply "infill" and not another type of "infill support". Thanks for your help!

Ghostkeeper, 2020-11-19:
This is indeed expected behaviour. It's the functionality of Infill Support that is supposed to do this in order to save printing time.

closed 2020-11-19 by Ghostkeeper