Ultimaker/Cura #8784: cura 4.8 no print preview

boyesreef, 2020-11-17:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-11-17:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

malwikl, 2020-11-21:
I face the same issues:
Application version
4.8.0

Platform
macOS Big Sur

Printer
Ender 3

Reproduction steps
Sometime after changing a setting, adding a custom support or any model related action and then slicing, the preview just presents an empty view. See screenshot

Screenshot(s)

[image]

Actual results
see above

Expected results
A normal slicing preview would be appreciated

Project file
Independet of project file used, had it even for a simple cube

Log file
'''
2020-11-21 10:55:11,624 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [264]: Going to slice build plate [0]!
2020-11-21 10:55:11,849 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [457]: Sending slice message took 0.2246990203857422 seconds
2020-11-21 10:55:11,944 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] Mesh has disconnected faces!
2020-11-21 10:55:11,952 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] Mesh has overlapping faces!
2020-11-21 10:55:11,961 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING]  -s coasting_enable="False" -s support_bottom_angles="[ ]" -s switch_extruder_extra_prime_amount="0" -s retraction_speed="25" -s infill_offset_y="0" -s prime_tower_size="20" -s draft_shield_dist="10" -s initial_extruder_nr="0" -s top_skin_preshrink="0.8" -s raft_base_line_width="0.8" -s bridge_sparse_infill_max_density="0" -s raft_interface_thickness="0.30000000000000004" -s line_width="0.4" -s support_zag_skip_count="10" -s material_no_load_move_factor="0.940860215" -s magic_mesh_surface_mode="normal" -s bridge_skin_material_flow="60" -s skin_angles="[ ]" -s center_object="False" -s material_linear_advance_factor_support="0.71" -s support_skip_zag_per_mm="20" -s fill_outline_gaps="False" -s connect_skin_polygons="False" -s infill_pattern="cubic" -s material_flow_temp_graph="[[3.5,200],[7.0,240]]" -s raft_interface_fan_speed="0" -s max_extrusion_before_wipe="10" -s raft_base_line_spacing="1.6" -s speed_support_infill="25.0" -s extruder_prime_pos_y="0" -s bottom_thickness="0.8" -s material_initial_print_temperature="210" -s material_break_retracted_position="-50" -s top_skin_expand_distance="0.8" -s machine_name="Creality Ender-3 Pro" -s material_linear_advance_factor_infill="0" -s bridge_enable_more_layers="True" -s skirt_brim_line_width="0.4" -s top_bottom_pattern="lines" -s cool_min_speed="10" -s raft_interface_acceleration="500" -s magic_fuzzy_skin_enabled="False" -s outer_inset_first="False" -s wireframe_bottom_delay="0" -s infill_line_width="0.4" -s magic_fuzzy_skin_outside_only="False" -s wall_line_count="2" -s z_seam_corner="z_seam_corner_weighted" -s machine_head_with_fans_polygon="[[-26, 34], [-26, -32], [32, -32], [32, 34]]" -s jerk_infill="8" -s speed_prime_tower="25.0" -s wireframe_flow_connection="100" -s mold_enabled="False" -s acceleration_support_interface="500" -s platform_adhesion="0" -s machine_heated_build_volume="False" -s raft_surface_thickness="0.2" -s skin_preshrink="0.8" -s speed_print="50.0" -s print_temperature="210" -s material_break_speed="25" -s speed_travel_layer_0="100.0" -s jerk_enabled="False" -s support_xy_distance="0.8" -s speed_wall="25.0" -s support_infill_rate="20" -s jerk_print="8" -s support_interface_skip_height="0.2" -s bridge_skin_speed_2="12.5" -s fill_perimeter_gaps="everywhere" -s layer_0_z_overlap="0.15" -s support_interface_offset="0.0" -s bridge_wall_material_flow="50" -s support_skip_some_zags="False" -s infill_support_angle="40" -s support_bottom_line_distance="2.4000240002400024" -s support_roof_density="33.333" -s travel="0" -s support_join_distance="2.0" -s wipe_hop_amount="0.2" -s skin_overlap="10.0" -s print_sequence="all_at_once" -s acceleration_layer_0="500" -s date="21-11-2020" -s support_bottom_offset="0.0" -s bridge_wall_speed="12.5" -s optimize_wall_printing_order="True" -s ironing_pattern="zigzag" -s raft_interface_line_width="0.8" -s travel_compensate_overlapping_walls_x_enabled="True" -s machine_steps_per_mm_z="50" -s material_print_temp_prepend="True" -s wireframe_printspeed_down="5" -s raft_base_jerk="8" -s travel_avoid_supports="True" -s acceleration_infill="500" -s support_extruder_nr="0" -s jerk_travel="8" -s support_roof_offset="0.0" -s jerk_skirt_brim="8" -s prime_tower_enable="False" -s mold_roof_height="0.5" -s retraction_retract_speed="25" -s retraction_min_travel="1.5" -s infill_multiplier="1" -s skirt_line_count="3" -s limit_support_retractions="True" -s speed_z_hop="5" -s material_final_print_temperature="210" -s conical_overhang_enabled="False" -s acceleration_roofing="500" -s retraction_hop_enabled="False" -s switch_extruder_retraction_speed="20" -s meshfix_maximum_deviation="0.025" -s acceleration_prime_tower="500" -s machine_acceleration="500" -s hole_xy_offset="0" -s wireframe_roof_inset="3" -s jerk_print_layer_0="8" -s material_anti_ooze_retracted_position="-4" -s wall_0_wipe_dist="0.0" -s material_linear_advance_factor_wall_0="0" -s ironing_line_spacing="0.1" -s small_feature_speed_factor="50" -s machine_nozzle_expansion_angle="45" -s support_enable="False" -s retraction_count_max="100" -s skin_line_width="0.4" -s material_flow_layer_0="100" -s wall_material_flow="100" -s retraction_combing="noskin" -s wall_overhang_speed_factor="100" -s jerk_support_infill="8" -s cool_fan_speed_min="100.0" -s wipe_move_distance="20" -s support_structure="normal" -s support_bottom_line_w [... cut ...]
2020-11-21 10:58:03,892 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-11-21 11:06:58,236 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token.
2020-11-21 11:06:58,474 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data
2020-11-21 11:07:08,913 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
'''

Additional information
(Extra information relevant to the issue.)

Ghostkeeper, 2020-11-26:
I think these are the relevant bits from that log then:

```
2020-11-21 10:55:18,077 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setVertexShader [121]: Vertex shader failed to compile: 
2020-11-21 10:55:18,082 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setFragmentShader [132]: Fragment shader failed to compile: 
2020-11-21 10:55:18,088 - ERROR - [MainThread] UM.View.GL.ShaderProgram.build [148]: Shader failed to link: 
```

Question is why the shaders fail to compile. It doesn't seem to give us any reason. We're printing the result of [QOpenGLShaderProgram.log()](https://doc.qt.io/qt-5/qopenglshaderprogram.html#log) there.

dertbv, 2020-12-13:
I am seeing this issue as well.

beezischillin, 2020-12-18:
This appears for me as well. I have Cura whitelisted in Little Snitch so it has access to the Internet. A few restarts make it work but then I start a new project and it fails again. It also generates invalid gcode when this happens.

mrx23dot, 2020-12-19:
How is the G code invalid? Isn't there a failsafe/sanity check against that? 
There should be zero possibility sending down invalid G code, printers are too expensive to break.

beezischillin, 2020-12-20:
I am a newbie so forgive me if I'm imprecise, but simply put It's blank. I put together my printer, downloaded some spacers from thingiverse and tried slicing them. I since managed to print them by restarting Cura a few times until it worked as it should.

When Cura starts up wrong and I can't interact with the models properly, clicking does nothing, I could only select anything from the object list and the tools didn't work, couldn't lay the model on a face or create blocker cubes and after slice the preview is empty and a message pops up that I haven't sliced anything. When I saved that and loaded it up in any gcode viewer there was nothing, if I loaded it on the printer it said target temp is 0 and the head never started moving.

When I can reproduce it I'll upload an example gcode

pzacjr, 2020-12-22:
I have the same problem using Cura 4.8 on Ultimaker S3 with macOS Big Sur and MacOS Catalina.
There are a few issues with Preview, Not being able to select objects, not being able to use the monitor.
They are all Cura 4.8 issues as Cura 4.7.1 do not seem to have these issues.
This does not look well for Cura nor Ultimaker on such an expensive machine as the S3.
Such a buggy piece of software, obviously not tested thoroughly and released prematurely.

danilo2311, 2020-12-23:
I have the same problem using Cura 4.8 on Ultimaker S3 with macOS Big Sur and MacOS Catalina. any solution ?

Ghostkeeper, 2020-12-23:
> This appears for me as well. I have Cura whitelisted in Little Snitch so it has access to the Internet. A few restarts make it work but then I start a new project and it fails again. It also generates invalid gcode when this happens.
> 
> When this happens I cannot select a model by clicking and cannot generate supports either (pressing the button does nothing)

I think you are running into a different issue here. This issue is pinned: https://github.com/Ultimaker/Cura/issues/8745
I'll mark these comments as off topic as to not distract any further from the original issue in this thread that Layer View is not rendering.

mrx23dot, 2021-01-08:
4.7.1 also fails to simulate on my Dell laptop too (that's the 3rd PC I try to make it work on):
[cura.log](https://github.com/Ultimaker/Cura/files/5789751/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/5789752/stderr.log)
I see layers, but not toolpath.
[image]

From log:

2021-01-07 19:10:25,359 - DEBUG - [MainThread] cura.CuraApplication.run [830]: Booting Cura took **40**.233147382736206 **seconds**
2021-01-07 19:10:27,469 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [210]: Support for Vertex Array Objects: True
2021-01-07 19:10:27,514 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2021-01-07 19:10:27,519 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.0.9902 Compatibility Profile Context
2021-01-07 19:10:27,523 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   ATI Technologies Inc.
2021-01-07 19:10:27,529 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: **ATI Mobility Radeon HD 5650**
2021-01-07 19:10:27,534 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
2021-01-07 19:10:53,641 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because **other side requested close.**
2021-01-07 19:10:56,992 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///D:/Progs/Cura/resources/qml/RoundedRectangle.qml:68: TypeError: **Cannot read property of null**
2021-01-08 23:17:09,971 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///D:/Progs/Cura/resources/qml/Preferences/ProfilesPage.qml:555: TypeError: **Cannot read property 'is_read_only' of null**
2021-01-08 23:17:28,238 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [60]: **Could not get metadata of container Unknown because it was not found.**
2021-01-08 23:17:28,279 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///D:/Progs/Cura/resources/qml/Preferences/Materials/MaterialsDetailsPanel.qml:91: **TypeError: Cannot read property 'container_node' of null**
D:\Progs\Cura\plugins\CuraEngineBackend\ProcessSlicedLayersJob.py:143: DeprecationWarning: The binary mode of fromstring is **deprecated**, as it behaves surprisingly on unicode inputs. Use frombuffer instead

revpar1980, 2021-01-12:
I am having the same issue with Cura 4.8.  I can;t see any layers in the preview tab.  I saw another thread that suggested the "Extruder 1" was not chanched and that was the problem.  That wasn;t my case but seemed to help a few others.

fvrmr, 2021-01-19:
@revpar1980 can you give us more information? Log file, OS?

vortechs2000, 2021-01-23:
I ran into the same issue of no preview rendering on a new MacBook Air with the M1 processor.  A workaround I found online was to use the "Force layer view compatibility mode" - after restarting, it worked, but was limited.  I wanted to post a log here, so I unchecked that option, restarted, and preview mode worked perfectly with all functionality.

So now I'm good to go on a M1-based MacBook Air running Big Sur 11.1

fvrmr, 2021-01-25:
@vortechs2000 it is a known issue that Cura won't run correctly on a MacBook with the M1 processor. 
Cura uses OpenGL libraries. With a recent update of the Big Sur OS, and on the new macbooks containing the M1 processing chips there are Apple has introduced incompatibilities between the OpenGL libraries used, and the latest firmware used by the Macbook.

This means that Ultimaker Cura is currently not supported on new Macbooks containing the M1 chip. Please bear with us while we look into fixing this in an upcoming Ultimaker Cura version.

mrx23dot, 2021-01-25:
3D object displaying still works great on old openGL, it's only the simulation that's problematic.

vortechs2000, 2021-01-26:
@fvrmr I know it's not supposed to work.  My point was that it *did* work when I turned on compatibility mode and then shut it back off.  It might be helpful for others on the M1 Macs as a workaround.

Xeno8888, 2021-02-01:
I have the same issue on a **non M1 MacBook**: MacBook Pro 15" 2018, macOS Big Sur 11.2
Cura: 4.8.0
Printer: Anycubic i3 Mega

I have no idea when this happens, but I think in this state I could also not select any model with my mouse pointer, just by right clicking and using "select all models" from context menu.

[image]

I saw some shader issues in the log and some Arcus errors, does this help? 
[cura.log](https://github.com/Ultimaker/Cura/files/5904461/cura.log)

`
2021-02-01 15:07:33,953 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2021-02-01 15:07:33,959 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.0s
2021-02-01 15:07:40,608 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PreviewStage
2021-02-01 15:07:40,636 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SimulationView
2021-02-01 15:07:40,637 - INFO - [MainThread] STLReader.STLReader.load_file [52]: Using NumPy-STL to load STL data.
2021-02-01 15:07:40,645 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.00043702125549316406 seconds
2021-02-01 15:07:40,646 - DEBUG - [MainThread] STLReader.STLReader._read [104]: Loaded a mesh with 12612 vertices
2021-02-01 15:07:40,652 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/SimulationView/simulationview_composite.shader]...
2021-02-01 15:07:40,658 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setVertexShader [121]: Vertex shader failed to compile:
2021-02-01 15:07:40,661 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setFragmentShader [132]: Fragment shader failed to compile:
2021-02-01 15:07:40,664 - ERROR - [MainThread] UM.View.GL.ShaderProgram.build [148]: Shader failed to link:
2021-02-01 15:07:40,751 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/color.shader]...
2021-02-01 15:07:40,763 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/SimulationView/layers3d.shader]...
2021-02-01 15:07:40,780 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/SimulationView/layers3d_shadow.shader]...
2021-02-01 15:07:40,791 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/toolhandle.shader]...
2021-02-01 15:07:40,797 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/color.shader]...
2021-02-01 15:07:40,803 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/striped.shader]...
2021-02-01 15:07:41,831 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PrepareStage
2021-02-01 15:07:41,835 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SolidView
2021-02-01 15:07:41,865 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2021-02-01 15:07:41,913 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/Menus/ConfigurationMenu/CustomConfiguration.qml:234: TypeError: Cannot read property 'activeStack'   of undefined
2021-02-01 15:07:41,927 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/xray.shader]...
2021-02-01 15:07:43,863 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [246]: Starting to slice...
2021-02-01 15:07:43,863 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [264]: Going to slice build plate [0]!
2021-02-01 15:07:43,869 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-02-01 15:07:43,974 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2021-02-01 15:07:44,012 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: /Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine
2021-02-01 15:07:44,016 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['/Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '']
2021-02-01 15:07:44,020 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend]

......

2021-02-01 15:11:22,607 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [252]: Changing perspective mode to orthographic.
2021-02-01 15:11:25,435 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2021-02-01 15:11:33,892 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [252]: Changing perspective mode to orthographic.
2021-02-01 15:11:44,058 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
[... cut ...]
2021-02-01 15:14:47,372 - DEBUG - [Thread-6] CuraEngineBackend.ProcessSlicedLayersJob.run [74]: Processing new layer for build plate 0...
2021-02-01 15:14:47,811 - DEBUG - [Thread-6] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 0.4386289119720459 seconds
2021-02-01 15:14:47,856 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [929]: See if there is more to slice(2)...
2021-02-01 15:14:58,081 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles`

Ghostkeeper, 2021-02-08:
Your issue is the same as the report from Malwikl then, Xeno8888. It's useful to know that it happens on multiple computers, but we're not yet at the bottom of this. We're planning on updating some of our dependencies to hopefully resolve this in the next release.

Ghostkeeper, 2021-02-24:
Possible similar report, but on Windows: https://github.com/Ultimaker/Cura/issues/9281

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Xeno8888, 2024-11-20:
> Is this still an issue in current Cura versions (5.8.0 and up)? Can this be closed?

I had no issues with the last releases.

closed 2024-11-20 by GregValiant