Ultimaker/Cura #8786: Cura 4.8 beta / 4.8

jarekdudzinski, 2020-11-17:
I'm having weird slicing results. All lines in this model are exactly 1.5mm width - modelled in Fusion. Tried my custom and default slicer settings with 0.5 line width.
Some of the lines are sliced as 2 external wall + 1 internal (as it should be), but some are having two very thin lines inside. This also generates weird fill on the corners, where the lines join.

[image]

jarekdudzinski, 2020-11-17:
PS: in Prusa slicer it looks like this (correct, I think):
[image]

konskarm, 2020-11-18:
Could you share the project file that reproduces this issue?

jarekdudzinski, 2020-11-18:
It's very simple - just 1.5mm lines intersecting. Attached below.
[Body7.stl.zip](https://github.com/Ultimaker/Cura/files/5561203/Body7.stl.zip)

Ghostkeeper, 2020-11-19:
Duplicate of https://github.com/Ultimaker/Cura/issues/6401.

Cura currently generates only even number of walls that loop around the border. Anything of exactly 1 line width or below is omitted and filled with gap filling instead (if enabled). A workaround is to reduce the line width by 0.001mm.
We also have a real fix in progress here: Ultimaker/CuraEngine#1210

closed 2020-11-19 by Ghostkeeper

jarekdudzinski, 2020-11-19:
Ok, there IS some improvement using 0.499mm line width - but intersections still looks strange, and because of it printing time is ~30-50% higher. Plus, you have many more additional retractions.
It generates those double-triangles (with small dots for filling them). 
[image]

Ghostkeeper, 2020-11-20:
Yeah, it will still print all walls as loops. So since they overlap by almost all of their width, you'd get overextrusion if Wall Overlap Compensation is not enabled. I'd also suggest to set Minimum Wall Flow to ~50% to prevent going over the same line twice with practically no flow. That would save a lot of time here.