Ultimaker/Cura #8787: [4.8.0] Gaps in shells

JonathanVerm, 2020-11-17:
**Application version**
Cura 4.8.0

**Platform**
Windows 10 
AMD Ryzen 5 4600H, 16GB RAM
GeForce GTX 1650, 4GB VRAM

**Printer**
Anycubic Predator

**Reproduction steps**
1. Loaded in the 3D model
2. Tested with other 3D models

**Screenshot(s)**
[image]

**Actual results**
The model gets slices, but there are several gaps in the shell.

**Expected results**
The shells should connect to create a closed perimeter.

**Project file**
[WeTransfer link](https://www.dropbox.com/t/Q11z9L47X8iZFRGa)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5555827/cura.log)

**Additional information**
-

Asterchades, 2020-11-18:
That's the coasting you've got enabled. It's set to 0.064mm³, which is going to be around a 0.8mm gap at every Z seam, or just over two of the 0.35mm walls you've got - which matches fairly well with the capture you've posted..

konskarm, 2020-11-18:
Well explained @Asterchades. I am closing this, since it is expected behavior.

closed 2020-11-18 by konskarm