Ultimaker/Cura #8789: latest version extremely slow

edlsmith, 2020-11-17:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
application hangs several times during session when moving from function to function. Issue not noticed in earlier releases of the qpplication.

jellespijker, 2020-11-18:
Can you fill in the template and provide log files

biovolt, 2020-11-29:
I have the same issue 
Just saving a project or slicing is way slower than the previous version
It sometimes stops responding and then crashes

nallath, 2020-12-01:
> I have the same issue
> Just saving a project or slicing is way slower than the previous version
> It sometimes stops responding and then crashes

The same holds for you. Knowing that it exists isn't enough for us to figure out what is going on. We really need the bug template to be filled in at the very least.

closed 2020-12-02 by no-response[bot]