Ultimaker/Cura #8790: Supports are not supporting some parts of file

Dogsrule333, 2020-11-18:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
Ender 5

**Reproduction steps**
1. Loaded file. Set print settings.  Sliced file.
2. Tried different files.

**Screenshot(s)**
[image]
[image]
[image]
[image]

**Actual results**
Most of the print looks correctly supported.  But there are areas that are not supported properly.  Either there are gaps, or support does not support anything.

**Expected results**
Supports should be correctly supporting areas that are needed

**Project file**
https://drive.google.com/drive/folders/11Lpp7YzTWaQj5V_s86hC5wxyuVbh9nAy?usp=sharing

**Log file**
No crashing issues

**Additional information**
I notice issue with tree and normal supports

Ghostkeeper, 2020-11-19:
Reduce the Minimum Support Area setting to get those bits supported. That's expected behaviour, just bad defaults.

The issue with tree support is a duplicate of this one: https://github.com/Ultimaker/Cura/issues/8622

Dogsrule333, 2020-11-20:
Thanks for the advice.  Sorry about the duplicate issue.
I tried adjusting the Minimum Support Area.  And did not see any difference.  The supports still have some odd placement issues.  Also The picture I included of the dragon miniature with the staff, still does not have support at the top of the staff, even with the Support area change.
I know you folks work hard at this.  So, Thank you for all you do.

Ghostkeeper, 2020-11-23:
For me it does support the top of the staff if the Minimum Support Area is set to 0:

[image]

Are you getting a different result?

The support is very small though because of the large Support X/Y Distance (and the general small scale of this print). Here are some settings that I think will work a bit better: 
[image]

Dogsrule333, 2020-11-25:
Thank you very much for your response.  I appreciate your thoughts.   I will try these changes out over the next couple of days and see how it goes.

I guess the thing that was puzzling me, was how it would put supports in, but they didn't seem to be touching anything, they were just floating there.  ex: top of dragon mini staff.  Similar thought to the cat miniature with the staff where the supports were not touching the majority of the staff.  But maybe I am over thinking things, and/or over reacting to gaps where I think things should be touching.

I will give the changes a go and see what happens.
Thanks again.  Have a good week.

Ghostkeeper, 2020-11-27:
Support sort of needs to be floating, at that scale, or you wouldn't be able to break it off. Unless it's soluble in water or something not chemically adhering to the build material, then it can be completely stuck to the model. Some Z distance is advisable if printing support with the same material as the model at least. But in your case I think the X/Y is a bit big, especially considering X/Y overrides Z. I normally prefer Z overrides X/Y for the Support Distance Priority setting.

closed 2020-12-11 by no-response[bot]